OP-TEE/build #559: QEMU build abruptly stops for 3.16.0

Hussain1811, 2022-04-19:
Steps to reproduce:

1)repo init -u https://github.com/OP-TEE/manifest.git -m qemu_v8.xml -b 3.16.0
2)repo sync -j4 --no-clone-bundle
3)cd build
4)make -j2 toolchains
5)make -j2 run

The VMs are fired up at all.
The build stops (without any error).

Last few set of logs:
```
  CC      drivers/tee/optee/smc_abi.o
  CC      drivers/interconnect/qcom/sdm845.o
  AR      drivers/interconnect/qcom/built-in.a
  CC      drivers/interconnect/core.o
  CC      drivers/tee/optee/ffa_abi.o
  CC      drivers/interconnect/bulk.o
  AR      drivers/tee/optee/built-in.a
  CC      drivers/tee/tee_core.o
  AR      drivers/interconnect/built-in.a
  CC      drivers/tee/tee_shm.o
  CC      drivers/tee/tee_shm_pool.o
  AR      drivers/tee/built-in.a
  AR      drivers/built-in.a
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.kallsyms1
  KSYMS   .tmp_vmlinux.kallsyms1.S
  AS      .tmp_vmlinux.kallsyms1.S
  LD      .tmp_vmlinux.kallsyms2
  KSYMS   .tmp_vmlinux.kallsyms2.S
  AS      .tmp_vmlinux.kallsyms2.S
  LD      vmlinux
  SORTTAB vmlinux
  SYSMAP  System.map
  OBJCOPY arch/arm64/boot/Image
make[1]: Leaving directory '/home/saddy/project/optee_qemu_3.16/linux'
```

Anything i am missing here?

jforissier, 2022-04-19:
Hi @Hussain1811,

You're probably missing the error due to the parallel build. When it fails please run "make" again without "-j2" so that the error will be at the end of the log.

Hussain1811, 2022-04-19:
Ahh - thanks @jforissier 
Looks like i have some issue in resolving proxies

jbech-linaro, 2022-04-19:
FYI, I just gave this a try and I had no issues, synced fine, built fine, run xtest without any issues.

Hussain1811, 2022-04-19:
I was able to build it successfully - the proxy was down.

Hussain1811, 2022-04-19:
By the way, where do i see the build logs of each component?

jforissier, 2022-04-19:
> By the way, where do i see the build logs of each component?

There is no log, if you want one you can `make 2>&1 | tee build.log`.

closed 2022-04-19 by Hussain1811