OP-TEE/build #568: Cannot build Qemu because `fcf-protection is not compatible with this target`

mrkct, 2022-05-24:
Hello, I followed the instructions at https://optee.readthedocs.io/en/latest/building/devices/qemu.html#qemu-v8 (qemu v8) and at the end when i run `make run` I got this message while building Qemu (i think?):  

```
[2577/2577] Linking target tests/qtest/qos-test
  AS      multiboot.o
  BUILD   multiboot.img
  BUILD   multiboot.raw
  SIGN    multiboot.bin
  AS      linuxboot.o
  BUILD   linuxboot.img
  BUILD   linuxboot.raw
  SIGN    linuxboot.bin
  CC      linuxboot_dma.o
cc1: error: ‘-fcf-protection’ is not compatible with this target
```

I have tried adding the `-fcf-protection=none` flag like this after removing the build directory (`rm -r ../qemu/build`):

```
diff --git a/common.mk b/common.mk
index 7635d30..88c4369 100644
--- a/common.mk
+++ b/common.mk
@@ -442,7 +442,7 @@ edk2-clean-common:
 ################################################################################
 # QEMU / QEMUv8
 ################################################################################
-QEMU_CONFIGURE_PARAMS_COMMON = --cc="$(CCACHE)gcc" --extra-cflags="-Wno-error" \
+QEMU_CONFIGURE_PARAMS_COMMON = --cc="$(CCACHE)gcc" --extra-cflags="-Wno-error fcf-protection=none" \
                               --disable-docs
 QEMU_EXTRA_ARGS +=\
        -object rng-random,filename=/dev/urandom,id=rng0 \
```
but it also fails with this message:
```
make[1]: Leaving directory '/home/marco/Desktop/op-tee/out-br'
make -C /home/marco/Desktop/op-tee/build/../linux LOCALVERSION= CROSS_COMPILE="/usr/bin/ccache /home/marco/Desktop/op-tee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" ARCH=arm64 Image scripts_gdb
make[1]: Entering directory '/home/marco/Desktop/op-tee/linux'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
make[1]: Leaving directory '/home/marco/Desktop/op-tee/linux'
mkdir -p /home/marco/Desktop/op-tee/build/../out/bin
ln -sf /home/marco/Desktop/op-tee/build/../linux/arch/arm64/boot/Image /home/marco/Desktop/op-tee/build/../out/bin
cd /home/marco/Desktop/op-tee/build/../qemu; ./configure --target-list=aarch64-softmmu\
                --cc="/usr/bin/ccache gcc" --extra-cflags="-Wno-error fcf-protection=none" --disable-docs
Using './build' as the directory for build output

ERROR: "/usr/bin/ccache gcc" either does not exist or does not work

make: *** [Makefile:232: /home/marco/Desktop/op-tee/build/../qemu/build/config-host.mak] Error 1
```

`/usr/bin/ccache gcc` does exist on my computer, I don't know what "does not work" means. This is the `config.log`:  https://gist.github.com/mrkct/c2f7a85e9109b9f01e74be5d80df0e42

I'm running Kubuntu 21.10. Any idea of what could be going wrong?

Thanks

jforissier, 2022-05-24:
Please try QEMU v7.0.0, it seems the issue has been fixed upstream by https://github.com/qemu/qemu/commit/9584d3d00a454f47b0341465142bcf0735d734ae.
```
$ cd qemu
$ git checkout v7.0.0
$ rm -rf build
$ cd ../build
$ make -j10 run
```

mrkct, 2022-05-24:
It worked! Small note: I also had to run `git fetch --all` in the qemu folder before

Thank you

closed 2022-05-24 by mrkct