OP-TEE/build #581: Package 'opensc' defined a second time from ../buildroot

jian3os, 2022-08-24:
jaloo@ECO-EV:~/backup/op-tee/src/build (master)$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   br-ext/Config.in
	deleted:    br-ext/package/opensc/Config.in
	deleted:    br-ext/package/opensc/opensc.mk

no changes added to commit (use "git add" and/or "git commit -a")
jaloo@ECO-EV:~/backup/op-tee/src/build (master)$ git diff br-ext/Config.in
diff --git a/br-ext/Config.in b/br-ext/Config.in
index 8b96f99..279d6ff 100644
--- a/br-ext/Config.in
+++ b/br-ext/Config.in
@@ -4,6 +4,5 @@ source "$BR2_EXTERNAL_OPTEE_PATH/package/optee_test_ext/Config.in"
 source "$BR2_EXTERNAL_OPTEE_PATH/package/optee_examples_ext/Config.in"
 source "$BR2_EXTERNAL_OPTEE_PATH/package/optee_rust_examples_ext/Config.in"
 source "$BR2_EXTERNAL_OPTEE_PATH/package/optee_benchmark_ext/Config.in"
-source "$BR2_EXTERNAL_OPTEE_PATH/package/opensc/Config.in"
 source "$BR2_EXTERNAL_OPTEE_PATH/package/ftpm_optee_ext/Config.in"
 source "$BR2_EXTERNAL_OPTEE_PATH/package/linux_ftpm_mod_ext/Config.in"

jforissier, 2022-08-24:
> Package 'opensc' defined a second time from ../buildroot

No, it's not:
```
$ ls -l ../buildroot/package/opensc
ls: cannot access '../buildroot/package/opensc': No such file or directory
```

jian3os, 2022-08-24:
That's my fault. 

according to the ../manifest/default.xml, buildroot should be version 2021.11

$ cd buildroot
$ git checkout -b 2021.11

it works well now, thanks for reply.

closed 2022-08-24 by jforissier