OP-TEE/build #582: Cannot build OP-TEE for Raspberry Pi 3

sepapple, 2022-08-25:
I'm trying to build OP-TEE for rpi3 , but there is the error which are not unzipped host-python3-setuptools 44.0.0 Extracting.

I followed the steps described in README.md and ran:
```
$ sudo apt install <packages described Prerequisites>
$ mkdir op-tee && cd op-tee
$ repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml
$ repo sync
$ cd build
$ make toolchains
$ make
```
I type ```/usr/bin/unzip -q -d /home/sepa/op-tee/out-br/build/host-python3-setuptools-44.0.0 /home/sepa/op-tee/buildroot/dl/python-setuptools/setuptools-44.0.0.zip``` directly into the terminal.
Then unzip process works fine, so I'm confused.

When I built OP-TEE for rpi3 before under similar conditions, I didn't encounter such this error and have a successful built. 

Does anyone happen to know what is missing and how to fix it?

- Environments
OS: Ubuntu 22.04.1 LTS

- Error log
```
>>> host-python3-setuptools 44.0.0 Extracting
mkdir -p /home/sepa/op-tee/out-br/per-package/host-python3-setuptools/host
mkdir -p /home/sepa/op-tee/out-br/per-package/host-python3-setuptools/target
/usr/bin/unzip -q -d /home/sepa/op-tee/out-br/build/host-python3-setuptools-44.0.0 /home/sepa/op-tee/buildroot/dl/python-setuptools/setuptools-44.0.0.zip
Archive:  /usr/bin/unzip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
note:  /usr/bin/unzip may be a plain executable, not an archive
unzip:  cannot find zipfile directory in one of /usr/bin/unzip or
        /usr/bin/unzip.zip, and cannot find /usr/bin/unzip.ZIP, period.
make[2]: *** [package/pkg-generic.mk:217: /home/sepa/op-tee/out-br/build/host-python3-setuptools-44.0.0/.stamp_extracted] error 9
make[1]: *** [Makefile:23: _all] error 2 
```

jbech-linaro, 2022-08-25:
Hi, I've just upgraded to 22.04 myself, let me also give it a try and if I can replicate is, let's see if we can find a fix for the issue.

sepapple, 2022-08-25:
Thank you.
One week ago, I was trying to build a development environment on Ubuntu 18.04, but the same error occurred.

Maybe, there is a problem with buildroot, not OS version dependencies.

jbech-linaro, 2022-08-25:
@sepapple , I've built it now and my build completed without build errors, the section where you're having an issue looks like this on my build

```bash
^[[3m>>> host-python3-setuptools 44.0.0 Extracting^[[23m                                                                                                                                                                                                                       
mkdir -p /home/jyx/devel/optee_projects/rpi3/out-br/per-package/host-python3-setuptools/host                                                                                                                                                                                   
mkdir -p /home/jyx/devel/optee_projects/rpi3/out-br/per-package/host-python3-setuptools/target                                                                                                                                                                                 
/usr/bin/unzip -q -d /home/jyx/devel/optee_projects/rpi3/out-br/build/host-python3-setuptools-44.0.0 /home/jyx/tmp/buildroot_dl/python-setuptools/setuptools-44.0.0.zip                                                                                                        
mv /home/jyx/devel/optee_projects/rpi3/out-br/build/host-python3-setuptools-44.0.0/setuptools-44.0.0/* /home/jyx/devel/optee_projects/rpi3/out-br/build/host-python3-setuptools-44.0.0                                                                                         
rm -f -r /home/jyx/devel/optee_projects/rpi3/out-br/build/host-python3-setuptools-44.0.0/setuptools-44.0.0                                                                                                                                                                     
^[[3m>>> host-python3-pycparser 2.20 Extracting^[[23m 
```

Searching for this on internet, says that it could be a corrupt zip file. But in your case you have verified that manually doing the operation everything works. So, I wonder if there perhaps is some race-condition happening. I.e., buildroot tries to unzip before the complete zip-file is available?

sepapple, 2022-08-25:
Now, I reinstall Ubuntu 22.04 LTS and can build OP-TEE correctly. 

The difference from the past is the type of OS.
When I failed to build, I used to use the Japanese Remix Ubuntu image.
The image changes unzip from the original image to correspond to Japanese file name.
By this change, I may have been failing building until now.

@jbech-linaro Thank you very much for your kind reply.

closed 2022-08-25 by sepapple