OP-TEE/build #589: edk2 fails - vla-parameter brotli error

jbech-linaro, 2022-09-13:
It seems like I've started to get the issue described here: https://github.com/patmagauran/i915ovmfPkg/issues/21 when trying to build OP_TEE for QEMU v8. In this case it seems to be something built for the host, i.e., not using the cross compiler. The version on my host PC is:

```bash
$ gcc --version
gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Based on the information at the link above I've played around with trying to disable when warning with by doing this in edk2. However, no success.

```diff
diff --git a/BaseTools/Conf/tools_def.template b/BaseTools/Conf/tools_def.template
index 498696e583..12f2aa3b2e 100755
--- a/BaseTools/Conf/tools_def.template
+++ b/BaseTools/Conf/tools_def.template
@@ -1863,7 +1863,7 @@ NOOPT_*_*_OBJCOPY_ADDDEBUGFLAG     = --add-gnu-debuglink=$(DEBUG_DIR)/$(MODULE_N
 *_*_*_DTCPP_PATH                   = DEF(DTCPP_BIN)
 *_*_*_DTC_PATH                     = DEF(DTC_BIN)
 
-DEFINE GCC_ALL_CC_FLAGS            = -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common
+DEFINE GCC_ALL_CC_FLAGS            = -g -Os -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -Wno-error=vla-parameter -include AutoGen.h -fno-common
 DEFINE GCC_IA32_CC_FLAGS           = DEF(GCC_ALL_CC_FLAGS) -m32 -malign-double -freorder-blocks -freorder-blocks-and-partition -O2 -mno-stack-arg-probe
 DEFINE GCC_X64_CC_FLAGS            = DEF(GCC_ALL_CC_FLAGS) -mno-red-zone -Wno-address -mno-stack-arg-probe
 DEFINE GCC_ARM_CC_FLAGS            = DEF(GCC_ALL_CC_FLAGS) -mlittle-endian -mabi=aapcs -fno-short-enums -funsigned-char -ffunction-sections -fdata-sections -fomit-frame-pointer -Wno-address -mthumb -mfloat-abi=soft -fno-pic -fno-pie
```

**Temporary fix**
Note that there is a fix [merged](https://github.com/google/brotli/pull/893). By applying that to brotli the error goes away.
```bash
$ cd <workspace>/edk2/BaseTools/Source/C/BrotliCompress/brotli
$ wget  wget https://github.com/google/brotli/pull/893.patch
$ git am 893.patch
$ cd <workspace>/build
$ make
```
Perhaps the solution is to step up our edk2 git so it contains this particular commit. If/when I have time, I'll probably look a bit more into this.

**Error message when compiling**
```bash
make[3]: Leaving directory '/home/jyx/devel/optee_projects/qemu_v8/edk2/BaseTools/Source/C/VfrCompile'
make -C BrotliCompress
make[3]: Entering directory '/home/jyx/devel/optee_projects/qemu_v8/edk2/BaseTools/Source/C/BrotliCompress'
gcc  -c -I ./brotli/c/include -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/ -MD -fshort-wchar -fno-strict-aliasing -fwrapv -fno-delete-null-pointer-checks -Wall -Werror -Wno-deprecated-declarations -Wno-stringop-truncation -Wno-restrict -Wno-unused-result -nostdlib -g -O2  BrotliCompress.c -o BrotliCompress.o
gcc  -c -I ./brotli/c/include -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/ -MD -fshort-wchar -fno-strict-aliasing -fwrapv -fno-delete-null-pointer-checks -Wall -Werror -Wno-deprecated-declarations -Wno-stringop-truncation -Wno-restrict -Wno-unused-result -nostdlib -g -O2  brotli/c/dec/decode.c -o brotli/c/dec/decode.o
brotli/c/dec/decode.c:2033:41: error: argument 2 of type ‘const uint8_t *’ {aka ‘const unsigned char *’} declared as a pointer [-Werror=vla-parameter]
 2033 |     size_t encoded_size, const uint8_t* encoded_buffer, size_t* decoded_size,
      |                          ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from brotli/c/dec/decode.c:7:
./brotli/c/include/brotli/decode.h:204:19: note: previously declared as a variable length array ‘const uint8_t[*decoded_size]’ {aka ‘const unsigned char[*decoded_size]’}
  204 |     const uint8_t encoded_buffer[BROTLI_ARRAY_PARAM(encoded_size)],
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
brotli/c/dec/decode.c:2034:14: error: argument 4 of type ‘uint8_t *’ {aka ‘unsigned char *’} declared as a pointer [-Werror=vla-parameter]
 2034 |     uint8_t* decoded_buffer) {
      |     ~~~~~~~~~^~~~~~~~~~~~~~
In file included from brotli/c/dec/decode.c:7:
./brotli/c/include/brotli/decode.h:206:13: note: previously declared as a variable length array ‘uint8_t[encoded_size]’ {aka ‘unsigned char[encoded_size]’}
  206 |     uint8_t decoded_buffer[BROTLI_ARRAY_PARAM(*decoded_size)]);
      |     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[3]: *** [../Makefiles/footer.makefile:21: brotli/c/dec/decode.o] Error 1
make[3]: Leaving directory '/home/jyx/devel/optee_projects/qemu_v8/edk2/BaseTools/Source/C/BrotliCompress'
make[2]: *** [GNUmakefile:76: BrotliCompress] Error 2
make[2]: Leaving directory '/home/jyx/devel/optee_projects/qemu_v8/edk2/BaseTools/Source/C'
make[1]: *** [GNUmakefile:19: Source/C] Error 2
make[1]: Leaving directory '/home/jyx/devel/optee_projects/qemu_v8/edk2/BaseTools'
make: *** [common.mk:423: edk2-common] Error 2
```

jforissier, 2022-09-14:
Strange, I cannot reproduce this although I have the exact same host compiler and have cloned `edk2/` from scratch again:
```
$ rm -rf edk2
$ repo sync edk2
$ cd edk2
$ git describe
edk2-stable201903-4029-gb24306f15d
$ git submodule update --init
$ cd ../build
$ gcc --version | head -1
gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
$ make edk2 V=1 2>&1 | tee edk2.log
...
- Done -
Build end time: 11:14:16, Sep.14 2022
Build total time: 00:00:41
$ vi edk2.log
...
gcc  -c -I ./brotli/c/include -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/ -MD -fshort-wchar -fno-strict-aliasing -fwrapv -fno-delete-null-pointer-checks -Wall -Werror -Wno-deprecated-declarations -Wno-stringop-truncation -Wno-restrict -Wno-unused-result -nostdlib -g -O2  brotli/c/dec/decode.c -o brotli/c/dec/decode.o
gcc  -c -I ./brotli/c/include -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/ -MD -fshort-wchar -fno-strict-aliasing -fwrapv -fno-delete-null-pointer-checks -Wall -Werror -Wno-deprecated-declarations -Wno-stringop-truncation -Wno-restrict -Wno-unused-result -nostdlib -g -O2  brotli/c/dec/huffman.c -o brotli/c/dec/huffman.o
...
```

jbech-linaro, 2022-09-14:
Weird, on my end it's 100% reproducible, only when applying the patch I'm able to get rid of it. Looks like more investigation is needed. Thanks for trying it out.

jbech-linaro, 2022-09-14:
I've tried the steps you've lined up and I crossed checked that I have the same commit. I still get the same error again.

jforissier, 2022-09-14:
Hmmmm we don't have the same source file it seems. Mine has the GCC 11.2.0 fix:

```
$ cd edk2/BaseTools/Source/C/BrotliCompress
$ sed -n '2033p' brotli/c/dec/decode.c
          s->dictionary->context_map[BROTLI_CONTEXT(p1, p2, s->context_lookup)]
$ cd brotli
$ git describe --tags
v1.0.9-35-gf4153a0
$ git log --oneline -n 5
f4153a0 (HEAD, origin/master, origin/HEAD, master) Fix for future versions of python (#911)
e83c7b8 Supress cmake warning (#931)
4ec6703 Merge pull request #929 from jbms/fix-vla-parameter
8376f72 Prepare for copybara (#939)
27dd726 Fix -Werror=vla-parameter errors with GCC 11.2.0
```

closed 2022-10-21 by github-actions[bot]