OP-TEE/build #592: AMD-Xilinx VCK190 build: verify_header failed

nathan-menhorn, 2022-09-27:
When executing the command `$ make -f versal.mk image` when following the instructions in:

https://github.com/OP-TEE/optee_docs/blob/master/building/devices/versal.rst#configuring-and-building-for-vck190 

an error is being encountered when trying to recreate this build using Ubuntu 20.04.5 LTS

`../u-boot/tools/mkimage -f versal/fitImage-versal-vck190.its versal/versal-vck190.ub`
`versal/fitImage-versal-vck190.its:17.20-19.15: Warning (unit_address_vs_reg): /images/kernel/hash@1: node has a unit name, but no reg property`
`versal/fitImage-versal-vck190.its:28.20-30.15: Warning (unit_address_vs_reg): /images/fdt-1/hash@1: node has a unit name, but no reg property`
`versal/fitImage-versal-vck190.its:39.20-41.15: Warning (unit_address_vs_reg): /images/fdt-2/hash@1: node has a unit name, but no reg property`
`versal/fitImage-versal-vck190.its:50.20-52.15: Warning (unit_address_vs_reg): /images/ramdisk/hash@1: node has a unit name, but no reg property`
`versal/fitImage-versal-vck190.its:57.16-65.11: Warning (unit_address_vs_reg): /configurations/conf@1: node has a unit name, but no reg property`
`versal/fitImage-versal-vck190.its:62.20-64.15: Warning (unit_address_vs_reg): /configurations/conf@1/hash@1: node has a unit name, but no reg property`
`../u-boot/tools/mkimage: verify_header failed for FIT Image support with exit code 1`
`make: *** [versal.mk:157: fitimage] Error 1`

The attached patch resolves the issue.

[0001-Removed-ats-to-avoid-verify_header-error.zip](https://github.com/OP-TEE/build/files/9659953/0001-Removed-ats-to-avoid-verify_header-error.zip)

cc: @ldts

jbech-linaro, 2022-09-28:
@nathan-menhorn , thanks for looking into this, would you please mind sending the fix as a regular pull request if it only affects the OP-TEE gits, which seems to be the case for this patch.

nathan-menhorn, 2022-09-28:
Sure thing.

closed 2022-09-30 by jforissier