OP-TEE/build #607: Add support building OP-TEE on the new Rockchip RK3588 (aka ROCK 5)?

TaihuLight, 2022-12-01:
@jforissier @jbech-linaro @jenswi-linaro @Dvergatal @petegriffin @JosephChen2017 @jonsmirl @ElonZhang0328
  [Trusted Substrate OpenEmbedded Layer](https://gitlab.com/Linaro/trustedsubstrate/meta-ts) works fine on Rock Pi 4B v1.5 with RK3399. It may also help to improve the building and running OP-TEE on the Rockchip series chips
The following issues may be helpful?
#https://github.com/OP-TEE/optee_os/issues/4197

The newest plat-rockchip: [RK3588](https://www.rutronik.com/fileadmin/Rutronik/Downloads/printmedia/products/01_semiconductors/Rockchip_Flyer_Microprocessor_Highlights_Online.pdf) also support ARM Trustzone. It is interesting and necessary to build and run OP-TEE on the new board [ROCK5](https://wiki.radxa.com/Rock5) with RK3588 for several developers and users of OP-TEE. However, 
**1 Does the current version of OP-TEE support RK3588?    
2 How to build it on Rock 5 running Ubuntu 20.04?**

[Rockchip_RK3588_Highlights.pdf](https://github.com/OP-TEE/optee_os/files/10128303/Rockchip_RK3588_Highlights.pdf)

jforissier, 2022-12-01:
> 1 Does the current version of OP-TEE support RK3588?

No. If you want to help support this board, you're welcome, please open pull requests in the `optee_os`, `build` and `manifest` Gits like I did for Rock Pi 4.

muvarov, 2022-12-01:
btw, use upstream link then my local copy. We moved here: https://gitlab.com/Linaro/trustedsubstrate/meta-ts

closed 2023-01-07 by github-actions[bot]

avi951, 2024-05-06:
Is there any solution of the above problem ?

NickDarvey, 2024-12-09:
I'm guessing it's closer now that https://github.com/OP-TEE/optee_os/pull/7059 is merged.

I guess we need these merged too:
 - https://github.com/edtubbs/manifest/tree/nanopc-t6
 - https://github.com/edtubbs/build/tree/nanopc-t6