OP-TEE/build #614: No Network Interface shown for Rock Pi board

prapty, 2023-01-24:
Hi, 
I have flashed the latest version of Op-TEE to a Rock Pi 4B+ board following the official documentation. The whole process worked without giving any errors and xtest was also successful after boot. 

**But the board does not show any network interface.** Executing _ifconfig_ I get only _auto lo_ interface. 

I tried to fix that by replicating the steps from [OP-TEE/build#386](https://github.com/OP-TEE/build/pull/386). However that did not work either.

Both ethernet and wifi work fine when Debian OS is run on the board. So the issue is not with hardware.

jforissier, 2023-01-25:
Hello @prapty,

Try `make LINUX_MODULES=y`.

prapty, 2023-01-27:
Hi @jforissier, 
I tried the command `make LINUX_MODULES=y`
Unfortunately, I am still getting only loopback interface when running `ifconfig`
```
# ifconfig -a
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```
However, I get the following output when running `ifup wlan0`
```
# ifup wlan0
Successfully initialized wpa_supplicant
Line 1: unknown global field 'ctrl_interface=/var/run/wpa_supplicant'.
Line 1: Invalid configuration line 'ctrl_interface=/var/run/wpa_supplicant'.
Failed to read or parse configuration '/etc/wpa_supplicant.conf'.
```
As I mentioned in the description of the issue, I followed the steps from [OP-TEE/build#386](https://github.com/OP-TEE/build/pull/386). So I manually added `br-ext/board/rockpi4/overlay/etc/network/interfaces` :

```
auto lo
iface lo inet loopback

iface wlan0 inet dhcp
        pre-up wpa_supplicant -i wlan0 -c /etc/wpa_supplicant.conf -B \
	       && sleep 6  # wait before starting udhcpc
        post-down killall -q wpa_supplicant
```
and `br-ext/board/rockpi4/overlay/etc/wpa_supplicant.conf`:
```
ctrl_interface=/var/run/wpa_supplicant

network={
  ssid="Network name"
  psk="Password"
}
```
I also modified `rockpi4.mk` and added the following lines:
```
BR2_PACKAGE_LINUX_FIRMWARE = y
BR2_PACKAGE_LINUX_FIRMWARE_TI_WL18XX = y
BR2_PACKAGE_WPA_SUPPLICANT = y
BR2_PACKAGE_WIRELESS_REGDB = y

BR2_ROOTFS_OVERLAY = $(ROOT)/build/br-ext/board/rockpi4/overlay
```
Lastly, I added following lines in `kconfigs/rockpi4.conf` 
```
CONFIG_CFG80211=y
CONFIG_MAC80211=y
CONFIG_RFKILL=y
CONFIG_WLCORE=y
CONFIG_WL18XX=y
CONFIG_WLCORE_SDIO=y
```
Please let me know if I am missing something.

prapty, 2023-02-13:
I figured out the solution for enabling network interfaces and specially wireless connection in Rock Pi 4 board. I am describing the fix here in case some one needs it in future:

1. Create `build/br-ext/board/rockpi4/overlay/etc/wpa_supplicant.conf` file with following contents:
```
country=Wi-Fi country code(i.e. US for United States)

update_config=1

network={
	ssid="wifi_network_ssid"
	scan_ssid=1
	psk="wifi_network_password"
	key_mgmt=WPA-PSK
}
```

2.  Create `build/br-ext/board/rockpi4/overlay/etc/network/interfaces` file with following contents:
```
auto lo
iface lo inet loopback
iface eth0 inet dhcp
iface wlan0 inet dhcp
        pre-up wpa_supplicant -i wlan0 -c /etc/wpa_supplicant.conf -B \
	       && sleep 6  # wait before starting udhcpc
        post-down killall -q wpa_supplicant
```

3. Extract attached compressed file and put following driver files in `build/br-ext/board/rockpi4/overlay/lib/firmware/brcm` folder:
```
brcmfmac43456-sdio.bin
brcmfmac43456-sdio.clm_blob
brcmfmac43456-sdio.radxa,rockpi4b.bin
brcmfmac43456-sdio.radxa,rockpi4b.txt
brcmfmac43456-sdio.txt
```

[brcm.zip](https://github.com/OP-TEE/build/files/10726918/brcm.zip)

These drivers are also available from [https://github.com/armbian/firmware](https://github.com/armbian/firmware)

4. Add following lines in `kconfigs/rockpi4.conf`
```
CONFIG_CFG80211=y
CONFIG_MAC80211=y
CONFIG_RFKILL=y
```

5. Add following lines in `rockpi4.mk`:
```
BR2_PACKAGE_LINUX_FIRMWARE = y
BR2_PACKAGE_WPA_SUPPLICANT = y
BR2_PACKAGE_WIRELESS_REGDB = y

BR2_ROOTFS_OVERLAY = $(ROOT)/build/br-ext/board/rockpi4/overlay
```
Change the value of `BR2_TARGET_ROOTFS_EXT2_SIZE` and set it to 2048M
`BR2_TARGET_ROOTFS_EXT2_SIZE = 2048M`

closed 2023-02-13 by prapty

jforissier, 2023-02-14:
@prapty thanks for the update. Still I can confirm I have no problem bringing up `eth0` with the upstream code, so there must be something going wrong on your side. Here are the commands I used.

```
$ mkdir rockpi4 && cd rockpi4
$ repo init -u https://github.com/OP-TEE/manifest -m rockpi4.xml
$ repo sync -j10
$ cd build
$ make -j2 toolchains
$ make -j$(noproc) BR2_TARGET_ROOTFS_EXT2_SIZE=384
$ sudo dd if=../out/rockpi4.img of=/dev/mmcblk0 && sudo sync /dev/mmcblk0
```
```
Welcome to Buildroot, type root or test to login
buildroot login: root
# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 5A:3C:5A:20:F6:FB
          inet addr:192.168.7.132  Bcast:192.168.7.255  Mask:255.255.255.0
[...]
```

(I will create a PR to increase the default root FS size to 384M, as the previous 256M are not enough now that the kernel has been updated)

geekaaron, 2025-08-09:
@jforissier May I ask which loader you are using? Here, I am using the document operation provided by optee for writing the rk3399_loader_v1.20.119.bin loader is stuck. Replace it after rk3399_loader_v1.27.126.bin, it can be written normally, but after rockpi4.img is burned in, there is no response. Other systems like ubuntu can start normally.