OP-TEE/build #616: opensc: package not present on 2021 buildroot

ldts, 2023-01-26:
Can we update to a more recent buildroot version?

I am not sure if this is a per board option but if it is I couldn't find it. 
I tested manually buildroot 2022.11.1 on Versal ACAP and opensc (pkcs11-tool) does show up.

jforissier, 2023-01-26:
Yes. In fact it was on my to-do list and I was waiting to see Xen updated to 4.16 or later in Buildroot, so that we can get rid of the patches in https://github.com/OP-TEE/build/tree/master/br-ext/patches/xen at the same time. But the BR package is still at 4.14.5 as I'm writing this (https://github.com/buildroot/buildroot/blob/master/package/xen/xen.hash).

So I guess an update to 2022.11.1 would be welcome at least for opensc.

ldts, 2023-01-27:
Ah great.
btw I couldnt figure out how to update this tree to upgrade buildroot directly hence why I didnt do a PR (for my testing I just git pulled buildroot after the first board build and rebuilt for my board again).

jforissier, 2023-01-27:
https://github.com/OP-TEE/manifest/pull/233

> (for my testing I just git pulled buildroot after the first board build and rebuilt for my board again)

That's perfectly fine and how I test things myself, too. To update the manifest you need to go into `.repo/manifests` and create a PR from that.

ldts, 2023-01-27:
argh of course. Soo many details...thanks!
one last thing before the weekend... the PKCS#11 TA is not in my buildroot in Versal. Would you know why?

jforissier, 2023-01-27:
You might need `CFG_PKCS11_TA=y` and possibly `CFG_USER_TA_TARGET_pkcs11=ta_arm64`, too.

closed 2023-01-30 by ldts