OP-TEE/build #624: CXX tests again failing to build with Arm gcc 11.3.1 release

b49020, 2023-02-08:
I was trying to uprev Arm gcc toolchain from 10.2 -> 11.3.1 but I have again hit CXX tests build issue with following error:

```
Building /home/sumit/build/upstream/optee_br/out-br/build/optee_examples_ext-1.0/random/ta/Makefile
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  CC      out/attestation.o
  CC      out/user_ta_header.o
  LD      out/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf
  CC      out/random_example_ta.o
  CC      out/user_ta_header.o
  LD      out/b6c53aba-9669-4668-a7f2-205629d00f86.elf
/home/sumit/build/upstream/optee_br/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld.bfd: /home/sumit/build/upstream/optee_br/toolchains/aarch64/bin/../lib/gcc/aarch64-none-linux-gnu/11.3.1/../../../../aarch64-none-linux-gnu/lib/../lib64/libstdc++.a(eh_alloc.o): in function `(anonymous namespace)::pool::free(void*) [clone .constprop.0]':
/data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-linux-gnu/obj/gcc3/aarch64-none-linux-gnu/libstdc++-v3/include/aarch64-none-linux-gnu/bits/gthr-default.h:749: undefined reference to `pthread_mutex_lock'
/home/sumit/build/upstream/optee_br/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld.bfd: /data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-linux-gnu/obj/gcc3/aarch64-none-linux-gnu/libstdc++-v3/include/aarch64-none-linux-gnu/bits/gthr-default.h:779: undefined reference to `pthread_mutex_unlock'
/home/sumit/build/upstream/optee_br/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld.bfd: /home/sumit/build/upstream/optee_br/toolchains/aarch64/bin/../lib/gcc/aarch64-none-linux-gnu/11.3.1/../../../../aarch64-none-linux-gnu/lib/../lib64/libstdc++.a(eh_alloc.o): in function `(anonymous namespace)::pool::allocate(unsigned long) [clone .constprop.0]':
/data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-linux-gnu/obj/gcc3/aarch64-none-linux-gnu/libstdc++-v3/include/aarch64-none-linux-gnu/bits/gthr-default.h:749: undefined reference to `pthread_mutex_lock'
/home/sumit/build/upstream/optee_br/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld.bfd: /data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-linux-gnu/obj/gcc3/aarch64-none-linux-gnu/libstdc++-v3/include/aarch64-none-linux-gnu/bits/gthr-default.h:779: undefined reference to `pthread_mutex_unlock'
/home/sumit/build/upstream/optee_br/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld.bfd: /home/sumit/build/upstream/optee_br/toolchains/aarch64/bin/../lib/gcc/aarch64-none-linux-gnu/11.3.1/libgcc_eh.a(unwind-dw2-fde-dip.o): in function `__gthread_mutex_lock':
/data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-linux-gnu/obj/gcc3/aarch64-none-linux-gnu/libgcc/./gthr-default.h:749: undefined reference to `pthread_mutex_lock'
/home/sumit/build/upstream/optee_br/build/../toolchains/aarch64/bin/aarch64-linux-gnu-ld.bfd: /home/sumit/build/upstream/optee_br/toolchains/aarch64/bin/../lib/gcc/aarch64-none-linux-gnu/11.3.1/libgcc_eh.a(unwind-dw2-fde-dip.o): in function `__gthread_mutex_unlock':
/data/jenkins/workspace/GNU-toolchain/arm-11/build-aarch64-none-linux-gnu/obj/gcc3/aarch64-none-linux-gnu/libgcc/./gthr-default.h:779: undefined reference to `pthread_mutex_unlock'
```

Any ideas @jforissier for a possible fix?

jforissier, 2023-02-08:
Perhaps libutils should provide pthread stubs as weak symbols?

jenswi-linaro, 2023-02-08:
The `libstdc++.a` usage has always been a bit suspect since it's compiled for a different environment. A proper fix would be to recompile it for a TA environment, basically a new toolchain target. That may be easier said than done though.

jforissier, 2023-02-08:
@jenswi-linaro agreed

b49020, 2023-02-09:
@jforissier Since you have better insights into C++ environment for TAs, would you be willing to take it up? Also, for the time being are you fine with `WITH_CXX_TESTS ?= n` for OP-TEE build with GCC v11.3.Rel1?

Actually, I have been playing around with rust support for OP-TEE and I am facing libc incompatibility issue with GCC 10.2 toolchain as follows:

```
# hello_world-rs 
hello_world-rs: /lib64/libc.so.6: version `GLIBC_2.33' not found (required by hello_world-rs)
```

jforissier, 2023-02-09:
> @jforissier Since you have better insights into C++ environment for TAs, would you be willing to take it up?

Sure.

> Also, for the time being are you fine with `WITH_CXX_TESTS ?= n` for OP-TEE build with GCC v11.3.Rel1?

Where would you like to change this? I'd rather not make any special case in the upstream code and fix the issues instead.

> Actually, I have been playing around with rust support for OP-TEE and I am facing libc incompatibility issue with GCC 10.2 toolchain as follows:
> 
> ```
> # hello_world-rs 
> hello_world-rs: /lib64/libc.so.6: version `GLIBC_2.33' not found (required by hello_world-rs)
> ```

This looks like a problem on the non-secure side (client app). I'd say the Rust runtime is linked against a newer version of libc than the one that is installed in the root FS by Buildroot. Either the `optee_rust` project uses its own toolchain instead of using the same as Buildroot (the `toolchains/aarch64` directory created by `make toolchains`), or (most probably) it downloads a Rust runtime already compiled with a newer toolchain (I doubt the Rust runtime on the non-secure side is built in the project; only the secure side for TAs). @DemesneGH

b49020, 2023-02-09:
> Where would you like to change this? I'd rather not make any special case in the upstream code and fix the issues instead.

I was just thinking to make a change here: https://github.com/OP-TEE/build/blob/master/common.mk#L51 but if you are already working on a fix then I can wait for the toolchain uprev.

> I'd say the Rust runtime is linked against a newer version of libc than the one that is installed in the root FS by Buildroot.

Yeah that should be the case but at this point I am a newcomer to rust so need to digest how all of the optee rust plumbing is done before I can actually contribute here.

DemesneGH, 2023-02-09:
> I'd say the Rust runtime is linked against a newer version of libc than the one that is installed in the root FS by Buildroot.

Strange. I believe the Rust examples use the same toolchains in OP-TEE repo and they're defined in there:

https://github.com/OP-TEE/build/blob/bcfa3f514ca403d8c8058e4eb937d2335b4d4fb6/common.mk#L299

And our CI works fine for gcc-10.2.

I'll try to reproduce the issue. What's the OP-TEE repo version you use?

b49020, 2023-02-09:
> I believe the Rust examples use the same toolchains in OP-TEE repo and they're defined in there

Just FYI I have rust compiler installed on my host machine as:

```
$ which rustc
/home/sumit/.cargo/bin/rustc
$ rustc --version
rustc 1.57.0-nightly (5ecc8ad84 2021-09-19)
```

> What's the OP-TEE repo version you use?

It the current master branch. I just followed this doc: https://optee.readthedocs.io/en/latest/building/optee_with_rust.html

DemesneGH, 2023-02-09:
@b49020 I didn't reproduce this error on version of the current master branch. Could you try to clean your building environment and rebuild the Rust examples?

jforissier, 2023-02-10:
@b49020 I have one good news and one bad news... which one do you want to hear first? :wink: :laughing: 

In fact this issue is not totally news to me because when I introduced the BTI/MTE/PAC CI job last week (https://github.com/OP-TEE/optee_os/pull/5805) I noticed the undefined symbols and tried a uClibc version of GCC. That avoided the undefined symbols but still failed to pass the C++ exception test (which I excluded with `-x regression_1031`). So here is the situation:

- The good news is that I have a fix for GCC 11.3.1. As I expected, stubbing `pthread_mutex_lock()` and `pthread_mutex_unlock()` in our `libutils` is enough to make things work again. I will post a PR soon.
- The bad news is that C++ support breaks again with GCC 12.2.0. This time there is a new undefined symbol: `_dl_find_object()`. And that one is much trickier it seems. It is a GNU extension to the dynamic loader functions in `<dlfnc.h>` (the traditional `dlopen()` etc.), which would probably be difficult to implement in OP-TEE. Note however that as I mentioned the uClibc version doesn't cause the undefined symbol (doesn't use `_dl_find_object()`) but still fails with a similar stack trace. So there may be hope still.

jforissier, 2023-02-10:
https://github.com/OP-TEE/optee_os/pull/5823

b49020, 2023-02-13:
Thanks @jforissier for working on this. I hope you will be able to resolve the pending issues.

b49020, 2023-02-13:
@DemesneGH

> Could you try to clean your building environment and rebuild the Rust examples?

I did rebuild everything from scratch but still same glibc runtime issue. I am using Ubuntu 22.04 as my host machine. Have you tested build on this distro? AFAICS, it looks like rust compiler maybe relying on cross compiler provided by the distro which is:

```
$ aarch64-linux-gnu-gcc --version
aarch64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
and hence rust client applications are linked against corresponding glibc.

DemesneGH, 2023-02-14:
@b49020 Please try this patch: https://github.com/OP-TEE/build/pull/626. Thanks.

b49020, 2023-02-14:
Thanks @DemesneGH, it did fixed the rust examples glibc incompatibility issue.

jforissier, 2023-03-17:
All issues mentioned here have been fixed, closing.

closed 2023-03-17 by jforissier