OP-TEE/build #628: failure to run with error "qemu-system-aarch64: -netdev user,id=vmnic: network backend 'user' is not compiled into this binary"

sammyne, 2023-02-15:
## Environment
- OP-TEE/build 3.20.0
- ubuntu 20.04 within VMware Workstation Pro 17

## Problem
Following the building instructions as [here](https://optee.readthedocs.io/en/latest/building/devices/qemu.html#qemu-v8), `make run` produces errors as
```bash
ln -sf /home/sammyne/workspaces/optee/build/../out-br/images/rootfs.cpio.gz /home/sammyne/workspaces/optee/build/../out/bin/

* QEMU is now waiting to start the execution
* Start execution with either a 'c' followed by <enter> in the QEMU console or
* attach a debugger and continue from there.
*
* To run OP-TEE tests, use the xtest command in the 'Normal World' terminal
* Enter 'xtest -h' for help.

cd /home/sammyne/workspaces/optee/build/../out/bin && /home/sammyne/workspaces/optee/build/../qemu/build/aarch64-softmmu/qemu-system-aarch64 \
	-nographic \
	-serial tcp:localhost:54320 -serial tcp:localhost:54321 \
	-smp 2 \
	-s -S -machine virt,secure=on,mte=off,gic-version=3,virtualization=false \
	-cpu max,pauth-impdef=on \
	-d unimp -semihosting-config enable=on,target=native \
	-m 1057 \
	-bios bl1.bin		\
	-initrd rootfs.cpio.gz \
	-kernel Image -no-acpi \
	-append 'console=ttyAMA0,38400 keep_bootcon root=/dev/vda2 ' \
	 \
	-object rng-random,filename=/dev/urandom,id=rng0 -device virtio-rng-pci,rng=rng0,max-bytes=1024,period=1000 -netdev user,id=vmnic -device virtio-net-device,netdev=vmnic
qemu-system-aarch64: -netdev user,id=vmnic: network backend 'user' is not compiled into this binary
make: *** [Makefile:434: run-only] Error 1
``` 

Really appreciate if someone could help~

jenswi-linaro, 2023-02-15:
I've heard of this problem from others too. It works on my system and in our CI so I can only assume that it's some prerequisite missing. Have you checked https://optee.readthedocs.io/en/latest/building/prerequisites.html#prerequisites ?

sammyne, 2023-02-15:
Based on [CHANGELOG](https://wiki.qemu.org/ChangeLog/7.2#Removal_of_the_%22slirp%22_submodule_(affects_%22-netdev_user%22)) of qemu 7.2, option `--enable-slirp` needs setting explicitly for building qemu.

My patch goes by adding these lines before https://github.com/OP-TEE/build/blob/master/common.mk#L471 as 
```makefile
# We need enable SLiRP explicitly since 7.2 according to 
#   https://wiki.qemu.org/ChangeLog/7.2#Removal_of_the_%22slirp%22_submodule_(affects_%22-netdev_user%22)
QEMU_CONFIGURE_PARAMS_COMMON += --enable-slirp
```
and install libslirp-dev for ubuntu 20.04.

Morever, https://optee.readthedocs.io/en/latest/building/prerequisites.html#prerequisites should add libslirp-dev for ubuntu 20.04 tab also.

@jenswi-linaro

jforissier, 2023-02-15:
> ```makefile
> # We need enable SLiRP explicitly since 7.2 according to 
> #   https://wiki.qemu.org/ChangeLog/7.2#Removal_of_the_%22slirp%22_submodule_(affects_%22-netdev_user%22)
> QEMU_CONFIGURE_PARAMS_COMMON += --enable-slirp
> ```

Is this really needed? The optee_os CI has the package installed (https://github.com/jforissier/docker_optee_os_ci/blob/60d8c0105fa2/Dockerfile#L79) but doesn't add `--enable-slirp`, yet everything works fine.

closed 2023-04-23 by github-actions[bot]