OP-TEE/build #630: How the memory size set in the qemu.mk file (such as 1057 at this time) is allocated to REE and TEE?

wsp1999, 2023-03-06:
.PHONY: run-only
run-only:
	$(call check-terminal)
	$(call run-help)
	$(call launch-terminal,54320,"Normal World")
	$(call launch-terminal,54321,"Secure World")
	$(call wait-for-ports,54320,54321)
	(cd $(BINARIES_PATH) && $(QEMU_PATH)/arm-softmmu/qemu-system-arm \
		-nographic \
		-serial tcp:localhost:54320 -serial tcp:localhost:54321 \
		-s -S -machine virt -machine secure=on -cpu cortex-a15 \
		-d unimp  -semihosting-config enable,target=native \
		-m 1057 \
		-bios $(ROOT)/out/bios-qemu/bios.bin \
		$(QEMU_EXTRA_ARGS) )

How the memory size set in the qemu.mk file (such as 1057 at this time) is allocated to REE and TEE?

jenswi-linaro, 2023-03-06:
OP-TEE and the Linux kernel reads this memory configuration from the DTB which QEMU generates when initializing. The value 1057 is mostly used for legacy reasons, it might be possible to use a more even value these days.

wsp1999, 2023-03-06:
Then, How do I change the memory ratio between TEE and REE?

jenswi-linaro, 2023-03-06:
The secure memory is using a fixed number hardcoded into QEMU, this is only non-secure memory. Unless you're using a static carveout for share memory everything is used by the REE.

wsp1999, 2023-03-06:
Does this means that 1057 is only non-secure memory? Where should I modify the fixed number to set the size of the safe memory?

jenswi-linaro, 2023-03-06:
It's hardcoded in `hw/arm/virt.c` in the QEMU source, search for VIRT_SECURE_MEM. If stuff breaks I'm not able to help.

wsp1999, 2023-03-20:
I tried to modify this value, the compilation passed normally, but there are always errors during the run. What is the reason for this?

wsp1999, 2023-03-22:
If I want to modify the size ratio of safe and non-safe memory in a project, what should I do？

closed 2023-04-27 by github-actions[bot]

n43e125, 2023-11-20:
I monitored that when setting memory larger than 1057 while having multiple core settings with -smp 4, it sometimes failed to bring up core 2,3,4. Are there any constraints on memory setting?