OP-TEE/build #649: [RFC] Makefiles for RISC-V platforms

gagachang, 2023-05-14:
Dear maintainers,

To let more RISC-V people build and try OP-TEE, it is helpful to have some Makefiles for RISC-V platforms.
Are there any recommendations for starting this work?

For example, add the following conditions into `common.mk` and `toolchain.mk`:
```
ifeq ($(ARCH), ARM)
...
else ifeq ($(ARCH), RISC-V)
...
endif
```

Or we should have another dedicated repository, such as `OP-TEE/build_riscv` ?

jenswi-linaro, 2023-05-15:
I believe keeping it all in the present build repository is preferable.

gagachang, 2023-05-15:
OK Thanks, I will prepare for it!

jenswi-linaro, 2023-05-15:
Yes, I guess we can do that. Please keep that in a separate commit.

gagachang, 2023-05-15:
> Yes, I guess we can do that. Please keep that in a separate commit.

Please check https://github.com/OP-TEE/build/pull/650.

gagachang, 2023-06-18:
Work in progress.

gagachang, 2023-07-19:
Please refer to https://github.com/OP-TEE/build/pull/666

gagachang, 2023-07-26:
Works on `common.mk`: https://github.com/OP-TEE/build/pull/669

gagachang, 2023-08-29:
Still WIP

closed 2023-10-04 by github-actions[bot]