OP-TEE/build #652: Fail to build OP-TEE 3.21.0 for Hikey 960

TaihuLight, 2023-05-20:
OP-TEE 3.20.0 can be built on my host PC successfully. But, OP-TEE 3.21.0 can not be built successfully on the same PC with the same environment. Errors shows as following:
`cc1: all warnings being treated as errors
make[1]: *** [GNUmakefile:489: /home/pelops2/hikey960/tee321/edk2/Build/HiKey960/RELEASE_GCC49/AARCH64/MdeModulePkg/Bus/Usb/UsbBusDxe/UsbBusDxe/OUTPUT/UsbBus.obj] Error 1
make[1]: Leaving directory '/home/pelops2/hikey960/tee321/edk2/Build/HiKey960/RELEASE_GCC49/AARCH64/MdeModulePkg/Bus/Usb/UsbBusDxe/UsbBusDxe'

build.py...
 : error 7000: Failed to execute command
	make tbuild [/home/pelops2/hikey970/tee321/edk2/Build/HiKey960/RELEASE_GCC49/AARCH64/MdeModulePkg/Bus/Usb/UsbBusDxe/UsbBusDxe]

build.py...
 : error F002: Failed to build module
	/home/pelops2/hikey970/tee321/edk2/MdeModulePkg/Bus/Usb/UsbBusDxe/UsbBusDxe.inf [AARCH64, GCC49, RELEASE]

- Failed -
Build end time: 08:56:56, May.20 2023
Build total time: 00:00:12

make: *** [Makefile:134: edk2] Error 1
`
@jforissier @jbech-linaro @jenswi-linaro

jforissier, 2023-06-10:
Try this:
```diff
diff --git a/hikey960.mk b/hikey960.mk
index dbff5c2..af167b3 100644
--- a/hikey960.mk
+++ b/hikey960.mk
@@ -134,7 +134,7 @@ edk2:
 		ln -s $(OPENPLATPKG_PATH)
 	set -e && cd $(EDK2_PATH) && source edksetup.sh && \
 		$(MAKE) -j1 -C $(EDK2_PATH)/BaseTools \
-			BUILD_CC="gcc $(call cc-option,gcc,-Wno-error=stringop-truncation,)" && \
+			BUILD_CC="gcc $(call cc-option,gcc,-Wno-error=stringop-truncation,) $(call cc-option,gcc,-Wno-error=stringop-overflow,)" && \
 		$(call edk2-call)
 
 .PHONY: edk2-cleandiff --git a/hikey960.mk b/hikey960.mk
index dbff5c2..af167b3 100644
--- a/hikey960.mk
+++ b/hikey960.mk
@@ -134,7 +134,7 @@ edk2:
 		ln -s $(OPENPLATPKG_PATH)
 	set -e && cd $(EDK2_PATH) && source edksetup.sh && \
 		$(MAKE) -j1 -C $(EDK2_PATH)/BaseTools \
-			BUILD_CC="gcc $(call cc-option,gcc,-Wno-error=stringop-truncation,)" && \
+			BUILD_CC="gcc $(call cc-option,gcc,-Wno-error=stringop-truncation,) $(call cc-option,gcc,-Wno-error=stringop-overflow,)" && \
 		$(call edk2-call)
 
 .PHONY: edk2-clean
```

EDK2 for HiKey is unmaintained, so unless someone takes up this task it is likely that sooner or later we will have to drop it and use something else (U-Boot), or even remove HiKey in OP-TEE support entirely.

TaihuLight, 2023-06-10:
@jforissier please do not remove the support for hikey960. It is important for reseachers.

jforissier, 2023-06-10:
> @jforissier I will try to do it tomorrow, and reply you timely. please do not remove OP-TEE support for hikey960. It is important for reseachers.

Don't worry, we will not remove anything unless there is a good reason and we will always discuss here first. I am glad to hear that people are still using HiKey, and I hope we can get patches to make it live longer! I personally used to use HiKey/HiKey960 a lot, but not so much so.

If my above suggestion works for you, please create a pull request so it can be merged (note that the `hikey.mk` file needs the same fix too I suppose). Thanks!

TaihuLight, 2023-06-11:
@jforissier 
(1) I am very sorry for that.  The method https://github.com/OP-TEE/build/issues/652#issuecomment-1585681108 dose not work. The same error occured.
(2) Besides, the secury memory size parameters including 
`CFG_TZDRAM_SIZE=0x04000000 CFG_TEE_RAM_VA_SIZE=0x00400000 CFG_DRAM1_BASE=0x430000000 CFG_SHMEM_SIZE=0x01800000  CFG_SHMEM_SIZE=0x00800000`
does not work. We must modify it in the source code 
`optee_os/core/arch/arm/plat-hikey/conf.mk` and `optee_os/core/arch/arm/plat-hikey/platform_config.h`
How to fix these two issues (1) and (2)?

jforissier, 2023-06-11:
In fact, the C flags have to be modified in `GCC_ALL_CC_FLAGS` in `edk2/BaseTools/Conf/tools_def.template` like so:
```diff
diff --git a/BaseTools/Conf/tools_def.template b/BaseTools/Conf/tools_def.template
index 2b4fb47ec..8a58208e8 100755
--- a/BaseTools/Conf/tools_def.template
+++ b/BaseTools/Conf/tools_def.template
@@ -4338,7 +4338,7 @@ DEBUG_*_*_OBJCOPY_ADDDEBUGFLAG     = --add-gnu-debuglink=$(DEBUG_DIR)/$(MODULE_N
 RELEASE_*_*_OBJCOPY_ADDDEBUGFLAG   =
 NOOPT_*_*_OBJCOPY_ADDDEBUGFLAG     = --add-gnu-debuglink=$(DEBUG_DIR)/$(MODULE_NAME).debug

-DEFINE GCC_ALL_CC_FLAGS            = -g -Os -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -include AutoGen.h -fno-common
+DEFINE GCC_ALL_CC_FLAGS            = -g -Os -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -Wno-stringop-overflow -include AutoGen.h -fno-common^M
 DEFINE GCC_IA32_CC_FLAGS           = DEF(GCC_ALL_CC_FLAGS) -m32 -malign-double -freorder-blocks -freorder-blocks-and-partition -O2 -mno-stack-arg-probe
 DEFINE GCC_X64_CC_FLAGS            = DEF(GCC_ALL_CC_FLAGS) -mno-red-zone -Wno-address -mno-stack-arg-probe
 DEFINE GCC_IPF_CC_FLAGS            = DEF(GCC_ALL_CC_FLAGS) -minline-int-divide-min-latency
```

Please refer to this Dockerfile, it worked for me: [Dockerfile.txt](https://github.com/OP-TEE/build/files/11715583/Dockerfile.txt)

Perhaps we should add the `sed` command to `hikey{,960}.mk`.

TaihuLight, 2023-06-12:
@jforissier Thank you very much.  The suggestion https://github.com/OP-TEE/build/issues/652#issuecomment-1586324847 is work for me. 
I would like to test OP-TEE support for Hikey 960 and then report the results. This work may be complemented in the later, because I must be buy a new UART-USB for testing the result.

closed 2023-07-18 by github-actions[bot]