OP-TEE/build #672: Error building op-tee for QEMUv8

Sourabh-ALTEN, 2023-07-27:
Hello,

I am trying to build the OP-TEE project for `QEMUv8`. My build setup is consists of WSL2 and docker container with Ubuntu 20.04. I have followed the [documentation](https://optee.readthedocs.io/en/latest/building/devices/qemu.html#qemu-v8) and steps

1. `make -j2 toolchains`
2. `make -j nproc`
3. `make run`

But during `make run` I am getting below error:
```
make[2]: Entering directory '/home/test/optee/qemu/build'
config-host.mak is out-of-date, running configure
bash: line 3: ./config.status: No such file or directory
make[2]: *** No rule to make target 'config-host.mak', needed by 'Makefile.prereqs'.  Stop.
make[2]: Leaving directory '/home/test/optee/qemu/build'
make[1]: *** [GNUmakefile:11: all] Error 2
make[1]: Leaving directory '/home/test/optee/qemu'
make: *** [Makefile:284: qemu] Error 2
```

Complete build log from `make 2>&1 | tee build.log` is here [build.log](https://pastebin.com/z8mLCSwG).

Can anyone please let me know what is the issue here and how can I resolve it?

Thanks in advance.

jforissier, 2023-07-27:
Try:
```
rm -rf ../qemu/build
make qemu
```

Sourabh-ALTEN, 2023-07-27:
Hi @jforissier thanks for quick response

`make qemu` seemed to work in the start

```
Found ninja-1.10.0 at /usr/bin/ninja
Running postconf script '/usr/bin/python3 /home/test/optee/qemu/scripts/symlink-install-tree.py'
make -C /home/test/optee/build/../qemu
make[1]: Entering directory '/home/test/optee/qemu'
changing dir to build for make ""...
make[2]: Entering directory '/home/test/optee/qemu/build'
/usr/bin/ninja  build.ninja && touch build.ninja.stamp
ninja: no work to do.
/usr/bin/python3 -B /home/test/optee/qemu/meson/meson.py introspect --targets --tests --benchmarks | /usr/bin/python3 -B scripts/mtest2make.py > Makefile.mtest
  GIT     ui/keycodemapdb meson tests/fp/berkeley-testfloat-3 tests/fp/berkeley-softfloat-3 dtc
  GIT     ui/keycodemapdb meson tests/fp/berkeley-testfloat-3 tests/fp/berkeley-softfloat-3 dtc
[1/2835] Generating config-poison.h with a custom command (wrapped by meson to capture output)
[2/2835] Generating qemu-version.h with a custom command (wrapped by meson to capture output)
[3/2835] Generating qemu-options.def with a custom command (wrapped by meson to capture output)
[4/2835] Generating qemu-img-cmds.h with a custom command (wrapped by meson to capture output)
[5/2835] Generating hmp-commands.h with a custom command (wrapped by meson to capture output)
[6/2835] Generating hmp-commands-info.h with a custom command (wrapped by meson to capture output)
.
.
```

After finishing 2835 tasks it continued with
```
make[2]: Leaving directory '/home/test/optee/qemu/build'
changing dir to build for make ""...
make[2]: Entering directory '/home/test/optee/qemu/build'
  GIT     ui/keycodemapdb meson tests/fp/berkeley-testfloat-3 tests/fp/berkeley-softfloat-3 dtc
[1/50] Generating qemu-version.h with a custom command (wrapped by meson to capture output)
[2/34] Generating tests/include/QAPI test (include) with a custom command
make[2]: Leaving directory '/home/test/optee/qemu/build'
make[1]: Leaving directory '/home/test/optee/qemu'
```
And then exited without any errors. Is this expected?

jforissier, 2023-07-27:
Yep that's fine, now try again `make run`...

Sourabh-ALTEN, 2023-07-28:
Now there seems to be an issue with `xterm`

```
make[1]: Entering directory '/home/test/optee/build'
ln -sf /home/test/optee/build/../out-br/images/rootfs.cpio.gz /home/test/optee/build/../out/bin/

* QEMU is now waiting to start the execution
* Start execution with either a 'c' followed by <enter> in the QEMU console or
* attach a debugger and continue from there.
*
* To run OP-TEE tests, use the xtest command in the 'Normal World' terminal
* Enter 'xtest -h' for help.

Warning: This program is an suid-root program or is being run by the root user.
The full text of the error or warning message cannot be safely formatted
in this environment. You may get a more descriptive message by running the
program as a non-root user or by removing the suid bit on the executable.
/usr/bin/xterm: Xt error: Can't open display: %s
/usr/bin/xterm: DISPLAY is not set
Warning: This program is an suid-root program or is being run by the root user.
The full text of the error or warning message cannot be safely formatted
in this environment. You may get a more descriptive message by running the
program as a non-root user or by removing the suid bit on the executable.
/usr/bin/xterm: Xt error: Can't open display: %s
/usr/bin/xterm: DISPLAY is not set
```

jforissier, 2023-07-28:
Run `./build/soc_term.py 54320` and `./build/soc_term.py 54321` in two separate windows before doing `make run` in the main one. Since you are using Docker you need to run these commands in the same container, that is, use `docker ps` to obtain the container ID then run `docker exec -it <container_id> /bin/bash`.

If you just want to run the OP-TEE tests in non-interactive mode, you may run `make check` instead of `make run`. No other terminal or `xterm` required in this case.

closed 2023-09-03 by github-actions[bot]