OP-TEE/build #676: Modify the linux kernel but not working

fengyuanyu1, 2023-08-24:
Hello, optee team,
   According to the [guideline](https://optee.readthedocs.io/en/latest/building/devices/qemu.html#qemu-v8), I build the Linux with optee in QEMU successfully. I want to add some functions in the linux kernel using optee. But I find that my modification in optee project doesn't work. My pipeline is as following:
```text
1. build optee project
2. modify the source code in <optee project>/linux
3. in <optee project>/build, make -f qemu_v8.mk linux
4. in <optee project>/build, make -f qemu_v8.mk run-only.
```
One of my modification is: 
```C
// <optee project>/linux/drivers/pci/probe.c
int __pci_read_base(struct pci_dev *dev, enum pci_bar_type type,
		    struct resource *res, unsigned int pos) {
        ...
        // line 318
        pci_info(dev, "Hack!!! reg 0x%x: %pR\n", pos, res);
        ...
}
```
And this modification can be complied successfully. But I cant find the HACK message in the boot log.
I try to recompile the whole project, the Linux, change the compile flag. But it is useless.

jforissier, 2023-08-24:
> 1. build optee project
> 2. modify the source code in <optee project>/linux
> 3. in <optee project>/build, make -f qemu_v8.mk linux
> 4. in <optee project>/build, make -f qemu_v8.mk run-only.

That's mostly good except that the linux kernel is now post-processed to produce uImage (to be loaded by U-Boot). So the "make linux" target does not do enough. Please try the following patch:
```diff
diff --git a/qemu_v8.mk b/qemu_v8.mk
index 92702e0..324f8cf 100644
--- a/qemu_v8.mk
+++ b/qemu_v8.mk
@@ -394,6 +394,8 @@ $(KERNEL_UIMAGE): u-boot linux | $(BINARIES_PATH)
 				-n "Linux kernel" \
 				-d $(BINARIES_PATH)/linux.bin $(KERNEL_UIMAGE)
 
+uImage: $(KERNEL_UIMAGE)
+
 $(ROOTFS_UGZ): u-boot buildroot | $(BINARIES_PATH)
 	ln -sf $(ROOT)/out-br/images/rootfs.cpio.gz $(BINARIES_PATH)
 	$(MKIMAGE_PATH)/mkimage -A arm64 \
```
...and do the following sequence instead:

1. build optee project
2. modify the source code in <optee project>/linux
3. in <optee project>/build, make uImage
4. in <optee project>/build, make run-only.

jforissier, 2023-08-24:
https://github.com/OP-TEE/build/pull/677

fengyuanyu1, 2023-08-24:
@jforissier Thanks for your quick reply. It works.

closed 2023-08-24 by fengyuanyu1