OP-TEE/build #680: Failure to run with manifest qemu_v8.xml and SPMC_AT_EL=2

mathieupoirier, 2023-08-31:
Hi OP-TEE people,

I am following the instructions found [here](https://optee.readthedocs.io/en/latest/building/devices/qemu.html#qemu-v8) except that I do "make SPMC_AT_EL=2 -j8 run".  I get the following after typing 'c' on the prompt:

`
QEMU 8.0.0 monitor - type 'help' for more information
(qemu) c
(qemu) write access to unsupported AArch64 system register op0:3 op1:0 crn:5 crm:6 op2:1
`

I get the exact same thing with QEMU 8.1.0.  What am I doing wrong?

Thanks,
Mathieu

jenswi-linaro, 2023-09-01:
Hi Mathieu,

That is one of the MTE system registers. Hafnium always expects MTE to be available so there should be a `mte=on` on the command line for QEMU. Can you see it?

By the way, when you've compiled the first time with either the "all" target or "run" target you can use the "run-only" target to start QEMU directly.

You could also try to remove the trusted-firmware-a/build/qemu directory in case you compiled once without `SPMC_AT_EL=2`. TF-A doesn't detect if compiler flags have changed.

Cheers,
Jens

mathieupoirier, 2023-09-04:
Hey Jens,

In fact I have been using `make run-only`.  In the original description of this issue, used an example that is as close as possible to what is found in the documentation to avoid misunderstandings.  Regarding, `mte=on`, you are 100% correct.  As can be seen [here](https://pastebin.com/Y21p6kqd) (line 5), the mte option is set incorrectly.  And that is because I did `make run-only` rather than `make SPMC_AT_EL=2 run-only`.   With the latter everything runs flawlessly.

Thanks for the insight,
Mathieu

closed 2023-09-04 by mathieupoirier