OP-TEE/build #700: Who can help me!make -f qemu.mk all and get a error :ccache: error: Exec format error

oldoldcoder, 2023-11-13:
I downloaded the 3.8.0 source code for op-tee through repo and used it in the build folder:
> Cd build
> Make - f toolchains. mk toolchains
To obtain the cross compilation tool, but I encountered an error using make - f qemu.mk all
[image]
This is my server information and information about the cross compilation tool
[image]
I realize this is an architectural issue, but I am unable to https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads Find a suitable cross compilation tool to re download to the toolchains folder.
I am a college student who is interested in op-tee and I hope to receive help. Thank you
**Basic: I cannot switch the architecture to Intel's servers**

jenswi-linaro, 2023-11-13:
If you use something more recent, preferably the latest release you'll have toolchains available for AArch64 hosts.

oldoldcoder, 2023-11-13:
> If you use something more recent, preferably the latest release you'll have toolchains available for AArch64 hosts.

my optee versioin now is 3.8.0，So I should download the latest version of OPTEE to solve this problem, right? I followed the instructions in a book step by step, so I didn't know what to do after the failure here. Thank you for your reply. I currently understand that if I modify my cross toolchain to use arm64, I may need to replace the makefile content, such as qemu.mk

oldoldcoder, 2023-11-13:
> If you use something more recent, preferably the latest release you'll have toolchains available for AArch64 hosts.

Should I use QEMU to simulate Intel architecture to run code

jforissier, 2023-11-13:
> > If you use something more recent, preferably the latest release you'll have toolchains available for AArch64 hosts.
> 
> my optee versioin now is 3.8.0，So I should download the latest version of OPTEE to solve this problem, right?

Yes, it would be the easiest option. You can find instructions at https://optee.readthedocs.io/en/latest/building/devices/qemu.html#id3 (64-bit OP-TEE for Armv8) and https://optee.readthedocs.io/en/latest/building/devices/qemu.html#build-instructions (32-bit-bit OP-TEE for Armv7). 

> Should I use QEMU to simulate Intel architecture to run code

No.

oldoldcoder, 2023-11-13:
> Yes, it would be the easiest option. You can find instructions at https://optee.readthedocs.io/en/latest/building/devices/qemu.html#id3 (64-bit OP-TEE for Armv8) and [...]

Thank you very much for your reply, it is very helpful for a beginner, wish you a happy life, I will try to use qemu_v8.xml next to get the code to run again

closed 2023-11-13 by oldoldcoder