OP-TEE/build #705: Openssl doesn't get built in when build called from qemu.mk

schnorea, 2023-11-26:
## Openssl doesn't seem to get built on unmodified build.
### Problem
After running `make qemu.mk` and successfully getting Normal and Secure World xterm windows. The Normal World doesn't seem to have `openssl` in fact all that can be found of `openssl` is
```bash
# find | grep -i openssl
./etc/ssl/openssl.cnf
./etc/ssl/openssl.cnf.dist
```
All other operations of the environment (as the tutorial in the docs mention) work as advertised.
### Desired State
My ultimate goal is to have optee as the keystore for openssl via pkcs11 to workout how this might be used for code signing, etc.
For this I will need at least to be able to access openssl.
### Context
This build environment is sourced from `optee/build` repo directly via this script (along with other elements)
```bash
#!/bin/bash
echo "This will take some time, since it's cloning lots of code ..."
yes "" | repo init -u https://github.com/OP-TEE/manifest.git
repo sync -j6
```
Make target: qemu.mk
Build environment: Docker container. - Build qemu.mk version of systems using this docker file 
https://github.com/jbech-linaro/docker_optee
### Attempted Debug
`common.mk` has openssl enabled
```bash
BR2_PER_PACKAGE_DIRECTORIES ?= y
BR2_PACKAGE_LIBOPENSSL ?= y
BR2_PACKAGE_MMC_UTILS ?= y
BR2_PACKAGE_OPENSSL ?= y
BR2_PACKAGE_OPTEE_BENCHMARK_EXT ?= $(CFG_TEE_BENCHMARK)
BR2_PACKAGE_OPTEE_BENCHMARK_EXT_SITE ?= $(BENCHMARK_APP_PATH)
BR2_PACKAGE_OPTEE_CLIENT_EXT_SITE ?= $(OPTEE_CLIENT_PATH)
BR2_PACKAGE_OPTEE_EXAMPLES_EXT ?= y
BR2_PACKAGE_OPTEE_EXAMPLES_EXT_CROSS_COMPILE ?= $(CROSS_COMPILE_S_USER)
BR2_PACKAGE_OPTEE_EXAMPLES_EXT_SDK ?= $(OPTEE_OS_TA_DEV_KIT_DIR)
BR2_PACKAGE_OPTEE_EXAMPLES_EXT_SITE ?= $(OPTEE_EXAMPLES_PATH)
```
Other elements from the config show up as would be expected but not `openssl`.

In looking through the build of the rootfs.cpio.gz
It looks like that config makes its way to the `out-br/defconfig` file via the python script
here. 
```bash
.PHONY: buildroot
buildroot: optee-os optee-rust
	@mkdir -p ../out-br
	@rm -f ../out-br/build/optee_*/.stamp_*
	@rm -f ../out-br/extra.conf
	@$(call append-br2-vars,../out-br/extra.conf)
	@(cd .. && $(PYTHON3) build/br-ext/scripts/make_def_config.py \
		--br buildroot --out out-br --br-ext build/br-ext \
		--top-dir "$(ROOT)" \
		--br-defconfig build/br-ext/configs/optee_$(BUILDROOT_ARCH) \
		--br-defconfig build/br-ext/configs/optee_generic \
		--br-defconfig build/br-ext/configs/$(BUILDROOT_TOOLCHAIN) \
		$(DEFCONFIG_GDBSERVER) \
		$(DEFCONFIG_XEN) \
		$(DEFCONFIG_TSS) \
		$(DEFCONFIG_TPM_MODULE) \
		$(DEFCONFIG_FTPM) \
		--br-defconfig out-br/extra.conf \
		--make-cmd $(MAKE))
	@$(MAKE) $(br-make-flags) -C ../out-br all
```
After this I lose my way in the  make started by the python script and the one that follows below it.

# Help
Any guidance or suggestion on why this shouldn't work and how to make it work welcome.

jenswi-linaro, 2023-11-27:
It looks like you need to set `BR2_PACKAGE_LIBOPENSSL_BIN=y` to get the openssl binary.

schnorea, 2023-12-01:
Thank you that did it!  How could I have discover this myself in the future for other libraries/utilities? Does `make menuconfig` highlight these dependencies?

closed 2023-12-12 by schnorea