OP-TEE/build #713: RPI3b:-sh: apt-get: not found

mlhh1103, 2023-12-19:
Hi, I power up the rpi3b, and login in as a root, but there is no sh file in the PATH, so "apt" and other commands such as "make" can not execute normally when I make the optee_client in rpi3, should I install some dependencies?  If so, what should I do?

jbech-linaro, 2023-12-19:
Although you in theory could do optee_client development on the RPi3 itself, that's not something that is supported in our RPi3 setup. I.e., you do all development on your host machine and then put the (cross)compiled binaries on the RPi3.

apt is a Debian package manager, the root fs in our Rpi3 setup is custom build using Buildroot. So if/when you need to add packages to the root fs, you have to enable them in the Buildroot configuration and rebuild. Remember, this is a development setup, which is heavily stripped down. It's not a normal Linux distro.

mlhh1103, 2023-12-21:
> Although you in theory could do optee_client development on the RPi3 itself, that's not something that is supported in our RPi3 setup. I.e., you do all development on your host machine and then put [...]

Oh, I see, should I bulid TEE-Supplicant in optee_client on the host machine, and put the compiled binaries on the RPI3 via NFS(which I already set)?

jbech-linaro, 2023-12-21:
> Oh, I see, should I bulid TEE-Supplicant in optee_client on the host machine, and put the compiled binaries on the RPI3 via NFS(which I already set)?

Yes, that's one way to do it. For people how hasn't configured NFS, they have to pull out the micro-SD card and repopulate the filesystem. Just make sure that you move all files produced when compiling optee_client, the library, tee-supplicant etc.

closed 2023-12-22 by mlhh1103