OP-TEE/build #723: Does the build support the config XEN_BOOT=y && SPCM_AT_EL=2 on qemu_v8

andyzsh68, 2024-01-22:
Hi, experts,
I want to confirm whether the build support the config XEN_BOOT=y && SPCM_AT_EL=2 on qemu_v8? In other words, non-secure virtualization and secure virtualization are both on simultaneously.
I have not  found any check against the configuration to say it is not supported, but when I build with the configuration with

make XEN_BOOT=y SPMC_AT_EL=2
make XEN_BOOT=y SPMC_AT_EL=2 run-only

I got errors complaining that mte is not support by qemu.

Can you help to confirm that?

thanks

jenswi-linaro, 2024-01-22:
I haven't tried that combination yet. I expect that some adjustments must be made before it works, for instance, use the `max` CPU instead of `cortex-a57`.

andyzsh68, 2024-01-24:
hi, jens,
Thanks for your reply.
Does this repo have plan to support this configuration? I think it is important to the new security arch(FF-A, SEL2, realm).
With this configuration, we have all the security roles, realm, non-secure, secure, root, all of them can communicate with each other.

Thanks

jenswi-linaro, 2024-01-24:
Yes, eventually we'll get there, it's a matter of prioritization. It would be very helpful to know which company you represent.

andyzsh68, 2024-01-24:
Hi, jens,
In fact, I represent no company but myself, just wanting to detailed understanding the new arm architecture features. In recent years, arm have published so many new features, and most of them are security related, the most effective way is to see how they work.
How can I subscribe the progress of the repo supported features(https://github.com/OP-TEE/manifest.git)?
Thanks

jenswi-linaro, 2024-01-24:
If you keep an eye on this repository you will be able to see how things progress.

closed 2024-01-25 by andyzsh68