OP-TEE/build #727: Rust error "could not rename downloaded file"

jforissier, 2024-01-29:
The Rust examples sometimes fail to build with "could not rename downloaded file" errors. Trying a second time "solves" the issue.

```
$ docker run -it --rm jforissier/optee_os_ci:qemuv8_check2
root@1c43ca64e015:/# cd ~
root@1c43ca64e015:~# ./get_optee_qemuv8.sh
[...]
root@1c43ca64e015:~# cd optee_repo_qemu_v8/build
root@1c43ca64e015:~# make -j128 check SPMC_AT_EL=2 2>&1 | tee build.log
[...]
error: component download failed for cargo-x86_64-unknown-linux-gnu: could not rename downloaded file from '/root/.rustup/downloads/fc0055ac726cf9fe06dce3bfac393b2744e7ff0970766d02414c1c3e28d0eefa.partial' to '/root/.rustup/downloads/fc0055ac726cf9fe06dce3bfac393b2744e7ff0970766d02414c1c3e28d0eefa'
```

Attaching [build.log](https://github.com/OP-TEE/build/files/14085561/build.log) (first build, with errors) and [build2.log](https://github.com/OP-TEE/build/files/14085567/build2.log) (second try immediately after the first one failed and without cleaning anything, OK).

b49020, 2024-02-05:
Should be fixed by: https://github.com/OP-TEE/build/pull/731

jforissier, 2024-02-08:
https://github.com/rust-lang/rustup/issues/988 might be relevant.

closed 2024-03-16 by github-actions[bot]