OP-TEE/build #736: Legacy BR options in tooclhain build

gyuri-szing, 2024-02-22:
Hi, `make toolchains` fails when executed on an aarch64 linux host. Buildroot complains about the configuration having legacy options. I have a fix for my environment but I am not sure if it would work for all possible use-cases. Also the aarch32 and riscv64 related files seem to have the same legacy values. I have no means to test these. 

Details about my environment:
 - `uname -a` on my machine returns: `Linux efe7e06cc6bf 5.15.0-94-generic #104-Ubuntu SMP Tue Jan 9 15:26:57 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux`
- I am using the current HEAD of master (ef9f8c2).
- My workspace is initialised with the repo tool using `repo init -u https://github.com/OP-TEE/manifest.git -b master -m fvp-ts.xml`

Build error:
```
gyoszi01@71fbd805e320:/src$ make -C build toolchains
make: Entering directory '/src/build'
Downloading gcc-arm-10.2-2020.11-aarch64-arm-none-linux-gnueabihf ...
Building aarch64 toolchain
make[1]: Entering directory '/src/buildroot'
mkdir -p /src/out-aarch64-sdk/build/buildroot-config/lxdialog
PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \
    obj=/src/out-aarch64-sdk/build/buildroot-config -C support/kconfig -f Makefile.br conf
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/src/out-aarch64-sdk/build/buildroot-config -DCONFIG_=\"\"  -MM *.c > /src/out-aarch64-sdk/build/buildroot-config/.depend 2>/dev/null || :
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/src/out-aarch64-sdk/build/buildroot-config -DCONFIG_=\"\"   -c conf.c -o /src/out-aarch64-sdk/build/buildroot-config/conf.o
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/src/out-aarch64-sdk/build/buildroot-config -DCONFIG_=\"\"  -I. -c /src/out-aarch64-sdk/build/buildroot-config/zconf.tab.c -o /src/out-aarch64-sdk/build/buildroo
t-config/zconf.tab.o
/usr/bin/gcc -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE  -I/src/out-aarch64-sdk/build/buildroot-config -DCONFIG_=\"\"   /src/out-aarch64-sdk/build/buildroot-config/conf.o /src/out-aarch64-sdk/build/buildroot-config/zconf
.tab.o  -o /src/out-aarch64-sdk/build/buildroot-config/conf
rm /src/out-aarch64-sdk/build/buildroot-config/zconf.tab.c
  GEN     /src/out-aarch64-sdk/Makefile
# 
# configuration written to /src/out-aarch64-sdk/.config
#
make[1]: Leaving directory '/src/buildroot'
make[1]: Entering directory '/src/out-aarch64-sdk'
Makefile.legacy:9: *** "You have legacy configuration in your .config! Please check your configuration.".  Stop.
make[1]: *** [Makefile:23: _all] Error 2
make[1]: Leaving directory '/src/out-aarch64-sdk'
make: *** [toolchain.mk:141: /src/build/../toolchains/aarch64/.done] Error 2
make: Leaving directory '/src/build'
gyoszi01@71fbd805e320:
``` 

The following patch solves the issue on my machine:
```
diff --git a/br-ext/configs/sdk-common b/br-ext/configs/sdk-common
index 3af1a37..d03f0f8 100644
--- a/br-ext/configs/sdk-common
+++ b/br-ext/configs/sdk-common
@@ -1,8 +1,8 @@
 BR2_CCACHE=y
 BR2_CCACHE_USE_BASEDIR=y
-BR2_GCC_VERSION_10_X=y
+BR2_GCC_VERSION_12_X=y
 BR2_INIT_NONE=y
-BR2_KERNEL_HEADERS_5_10=y
+BR2_KERNEL_HEADERS_6_1=y
 BR2_PACKAGE_BUSYBOX=n
 BR2_SYSTEM_BIN_SH_NONE=y
 BR2_TARGET_ROOTFS_TAR=n
diff --git a/br-ext/configs/toolchain-aarch64-sdk b/br-ext/configs/toolchain-aarch64-sdk
index 3e6ef34..342b6a8 100644
--- a/br-ext/configs/toolchain-aarch64-sdk
+++ b/br-ext/configs/toolchain-aarch64-sdk
@@ -2,10 +2,10 @@ BR2_TOOLCHAIN_EXTERNAL=y
 BR2_TOOLCHAIN_EXTERNAL_CUSTOM=y
 BR2_TOOLCHAIN_EXTERNAL_PATH="%TOP_DIR%/toolchains/aarch64"
 BR2_TOOLCHAIN_EXTERNAL_CUSTOM_PREFIX="aarch64-linux"
-BR2_TOOLCHAIN_EXTERNAL_GCC_10=y
-BR2_TOOLCHAIN_EXTERNAL_HEADERS_5_10=y
+BR2_TOOLCHAIN_EXTERNAL_GCC_12=y
+BR2_TOOLCHAIN_EXTERNAL_HEADERS_6_1=y
 BR2_TOOLCHAIN_EXTERNAL_CUSTOM_GLIBC=y
 BR2_TOOLCHAIN_EXTERNAL_CXX=y
-BR2_TOOLCHAIN_EXTERNAL_FORTRAN=y
-BR2_TOOLCHAIN_EXTERNAL_OPENMP=y
+BR2_TOOLCHAIN_EXTERNAL_FORTRAN=n
+BR2_TOOLCHAIN_EXTERNAL_OPENMP=n
 BR2_TOOLCHAIN_EXTERNAL_INET_RPC=n
```

jenswi-linaro, 2024-02-26:
I wonder if this relates to the recently updated buildroot? Anyway, would you mind creating a pull request to fix the issue?

gyuri-szing, 2024-02-27:
> I wonder if this relates to the recently updated buildroot? Anyway, would you mind creating a pull request to fix the issue?

Yes, that seems to be the case. I am happy to open a PR but I cannot test the aarch32 and riscv64 hosts. Is it ok to leave these unfixed?

jenswi-linaro, 2024-02-28:
They likely need the same changes so please make those too. Perhaps someone can help with the testing else I think it's OK to merge untested changes for `aarch32` and `riscv64`.

jforissier, 2024-02-28:
> > I'm fine with the moving GCC to 12 but OP-TEE build currently uses GCC 11.
> 
> We'll have issues the C++ support though? At least I don't think we've addressed that yet (some more info provided [here](https://bugs.linaro.org/show_bug.cgi?id=6015#c1)).

Yes, I'd like to keep the changes localized to the aarch64 host environment if possible, but that seems to be the case with the proposed patch.

PR #737 (2024-02-29): Fix Buildroot toolchain configuration files
The 2023.02 Buildroot update broke the aarch64, aarch32 and riscv64 toolchain external packages (br-ext). Buildroot complains about legacy options being in the configuration.
This patch removes the legacy values and fixes the toolchain build.

This PR fixes #736