OP-TEE/build #744: optee_test_ext: /bin/bash: Argument list too long

imre-kis-arm, 2024-04-16:
The hooks in [optee_test_ext.mk](https://github.com/OP-TEE/build/blob/6b9359d7933485fd8c01b391f6f04c8582d911f9/br-ext/package/optee_test_ext/optee_test_ext.mk) chains each TA's build step into a single command (note `&&` at the end of the foreach body). `$(TARGET_CONFIGURE_OPTS)` contains the whole environment, including `PATH` and the absolute path of each compiler tool (CC, LD, AS, OBJCOPY, etc.). This causes the hook to generate a command which is around 100k long (checked by removing `@` before the foreach). Depending on the length of the build directory path and the system's limitation on the maximal command length this can cause a build failure.

```
[100%] Built target xtest
/opt/cmake-3.18.4-Linux-x86_64/bin/cmake -E cmake_progress_start /[...]/optee-spmc-test/out-br/build/optee_test_ext-1.0/CMakeFiles 0
make[2]: /bin/bash: Argument list too long
make[2]: *** [package/pkg-generic.mk:284: /[...]/optee-spmc-test/out-br/build/optee_test_ext-1.0/.stamp_built] Error 127
make[1]: *** [Makefile:23: _all] Error 2
make[1]: Leaving directory '/[...]/optee-spmc-test/out-br'
make: *** [common.mk:341: buildroot] Error 2
make: Leaving directory '/[...]/optee-spmc-test/build'
```

It would be better to run each TA build as a separate command or have separate hooks for each TA if that's possible.

Although I didn't have problems with [optee_examples_ext](https://github.com/OP-TEE/build/blob/6b9359d7933485fd8c01b391f6f04c8582d911f9/br-ext/package/optee_examples_ext/optee_examples_ext.mk), it uses the same pattern, so it might worth changing that too.

Tested with 4.2.0.

jforissier, 2024-04-17:
Hi @imre-kis-arm,

The build process for TAs in `optee_test_ext.mk` is quite convoluted, in order to address the issue I decided to delegate the build to the makefile in `optee_test/ta`. Please see https://github.com/OP-TEE/optee_test/pull/740 which is a prerequisite then https://github.com/OP-TEE/build/pull/745.

imre-kis-arm, 2024-04-18:
Hi @jforissier,

I ran the tests which previously failed due to the "Argument list too long" issue and now they work fine, so thank you for the fix. I used the 4.2.0 components and the mentioned PRs (build: 758aaeb, optee_test: fdcbd69).

imre-kis-arm, 2024-04-23:
After running the test using the master branches, I can confirm that the issue is resolved, so I am closing this ticket.
Thank you for fixing it.

closed 2024-04-23 by imre-kis-arm