OP-TEE/build #751: C++ tests fail on aarch64 hosts

msgilligan, 2024-05-21:
Note: This is a new sub-issue from Issue #747 that addresses one of several problems that occur when trying to build on an aarch64 host.

When building (`cd build; make check`) on an aarch64 host with the current `master` branch, the following error occurs:

```
/usr/bin/ccache /optee/build/../toolchains/aarch64/bin/aarch64-linux-ld.bfd -e__ta_entry -pie -T out/ta/os_test/ta.lds -Map=out/ta/os_test/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.map --sort-section=alignment -z max-page-size=4096  --as-needed   --dynamic-list out/ta/os_test/dyn_list --eh-frame-hdr out/ta/os_test/init.o out/ta/os_test/os_test.o out/ta/os_test/ta_entry.o out/ta/os_test/test_float_subj.o out/ta/os_test/cxx_tests.o out/ta/os_test/cxx_tests_c.o out/ta/os_test/attestation.o out/ta/os_test/user_ta_header.o -L/optee/out-br/build/optee_test_ext-1.0/ta/os_test_lib/out/ta/os_test_lib -los_test -ldl -L/optee/optee_os/out/arm/export-ta_arm64/lib --start-group -lutils -lutee -lmbedtls -ldl /optee/toolchains/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/12.3.0/../../../../aarch64-buildroot-linux-gnu/lib/../lib64/libstdc++.a /optee/toolchains/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/12.3.0/libgcc_eh.a --end-group /optee/toolchains/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/12.3.0/libgcc.a -lutils -o out/ta/os_test/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf
/optee/build/../toolchains/aarch64/bin/aarch64-linux-ld.bfd: /optee/toolchains/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/12.3.0/libgcc_eh.a(unwind-dw2-fde-dip.o): in function `_Unwind_Find_FDE':
/optee/out-aarch64-sdk/build/host-gcc-final-12.3.0/build/aarch64-buildroot-linux-gnu/libgcc/../../../libgcc/unwind-dw2-fde-dip.c:512: undefined reference to `_dl_find_object'
make[4]: *** [/optee/optee_os/out/arm/export-ta_arm64/mk/link.mk:123: out/ta/os_test/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf] Error 1
make[3]: *** [/optee/out-br/build/optee_test_ext-1.0/ta/Makefile.gmake:61: ta-os_test] Error 2
make[2]: *** [package/pkg-generic.mk:284: /optee/out-br/build/optee_test_ext-1.0/.stamp_built] Error 2
make[1]: *** [Makefile:23: _all] Error 2
make[1]: Leaving directory '/optee/out-br'
make: *** [common.mk:341: buildroot] Error 2
```

This issue can be avoided with the workaround of setting `WITH_CXX_TESTS=n` on the `make` command line.

Note that with the current `master` branch to get this far you need to work around Issue #749, by either setting `RUST_ENABLE=n` or by applying the patch in PR #748 to `toolchain.mk`.

For additional history, see:

* https://github.com/OP-TEE/build/issues/747#issuecomment-2110273904
* https://github.com/OP-TEE/build/issues/747#issuecomment-2110509836
* https://github.com/OP-TEE/build/issues/747#issuecomment-2123573751

The current `toolchain.mk` does not download a toolchain for aarch64 on aarch64 hosts, but relies upon `buildroot` to build one, and it seems that `buildroot` is building the wrong version (see https://github.com/OP-TEE/build/issues/747#issuecomment-2123573751 for details)

So it seems there are at least three possible solutions:

1. Find a pre-built compatible toolchain somewhere and download it
2. Update the build to use a newer toolchain -- such as 13.2.Rel1, available on the [Arm GNU Toolchain Downloads](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) page.
3. Configure `buildroot` to build a compatible version

I am new to OP-TEE, these toolchains, and to buildroot so I'm definitely looking for advice on how to proceed. If I do find a solution I will submit a PR.

jforissier, 2024-05-22:
Hi @msgilligan, thanks for the detailed description. Could you please try https://github.com/OP-TEE/build/pull/753?

msgilligan, 2024-05-23:
Fixed by #753

closed 2024-05-23 by msgilligan