OP-TEE/build #759: Differences Between Using Buildroot and OP-TEE Build for OP-TEE Components

kewer127, 2024-06-09:
Hi developers,

I found that Buildroot supports the following OP-TEE components:

```
boot/optee-os/
boot/optee-os/optee-os.hash
boot/optee-os/optee-os.mk
package/optee-client/
package/optee-client/optee-client.hash
package/optee-client/optee-client.mk
package/optee-examples/
package/optee-examples/optee-examples.hash
package/optee-examples/optee-examples.mk
package/optee-test/
package/optee-test/optee-test.hash
package/optee-test/optee-test.mk
```

I am wondering what the difference is when using buildroot and this repository.

jbech-linaro, 2024-06-10:
It's the same as we have here, it was @etienne-lms who did upstream support in Buildroot to be able to build the ones you listed directly from Buildroot. The official setup you find in the OP-TEE documentation is slighly different, since we clone and work with the OP-TEE gits as standalone gits.

kewer127, 2024-06-11:
@jbech-linaro  Thanks for clarifying!

closed 2024-06-11 by kewer127