OP-TEE/build #763: /bin/bash: line 1: /.cargo/env: No such file or directory

ikitayama, 2024-06-18:
I am trying to build QEMU v8 in Ubuntu VM on M1 MAX. `make toolchains` seems to have built okay, but `make run` ended up with an error:

```
# Ensure the toolchain, components, and targets we've specified in
# rust-toolchain.toml are ready to go. Since that file sets rustup's
# default toolchain for the entire directory, all we need to do is run
# any rustup-wrapped command to trigger installation. We've arbitrarily
# chosen "cargo --version" since it has no other effect.
Configuring OP-TEE rust examples
/bin/bash: line 1: /.cargo/env: No such file or directory
make[2]: *** [package/pkg-generic.mk:273: /home/itaru/work/out-br/build/optee_rust_examples_ext-1.0/.stamp_configured] Error 1
make[1]: *** [Makefile:23: _all] Error 2
make[1]: Leaving directory '/home/itaru/work/out-br'
make: *** [common.mk:341: buildroot] Error 2
itaru@vm3:build$ cargo --version
cargo 1.75.0
```

I'm on branch 4.2.0 and using the upstream's qemu_v8.xml file.

jforissier, 2024-06-18:
Hi @ikitayama,

This is a known issue, see https://github.com/OP-TEE/build/issues/747#issuecomment-2108629173. TL;DR: use `make -j$(nproc) RUST_ENABLE=n`.

If you can find a way to make it work you're welcome to submit pull requests. In the mean time I think we should force `RUST_ENABLE=n` on non-x64_64 hosts. I'll make a PR for that.

Thanks!

ikitayama, 2024-06-18:
> Hi @ikitayama,
> 
> This is a known issue, see [#747 (comment)](https://github.com/OP-TEE/build/issues/747#issuecomment-2108629173). TL;DR: use `make -j$(nproc) RUST_ENABLE=n`.
> 
> If you can find a way to make it work you're welcome to submit pull requests. In the mean time I think we should force `RUST_ENABLE=n` on non-x64_64 hosts. I'll make a PR for that.
> 
> Thanks!

@jforissier I get:

```
[100%] Built target xtest
Building /home/itaru/work/out-br/build/optee_test_ext-1.0/ta/os_test_lib/Makefile
make[3]: Nothing to be done for 'all'.
Building /home/itaru/work/out-br/build/optee_test_ext-1.0/ta/os_test_lib_dl/Makefile
make[3]: Nothing to be done for 'all'.
Building /home/itaru/work/out-br/build/optee_test_ext-1.0/ta/os_test/Makefile
  GEN     out/dyn_list
  LD      out/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf
/home/itaru/work/build/../toolchains/aarch64/bin/aarch64-linux-ld.bfd: /home/itaru/work/toolchains/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/12.3.0/libgcc_eh.a(unwind-dw2-fde-dip.o): in function `_Unwind_Find_FDE':
/home/itaru/work/out-aarch64-sdk/build/host-gcc-final-12.3.0/build/aarch64-buildroot-linux-gnu/libgcc/../../../libgcc/unwind-dw2-fde-dip.c:512: undefined reference to `_dl_find_object'
make[3]: *** [/home/itaru/work/optee_os/out/arm/export-ta_arm64/mk/link.mk:123: out/5b9e0e40-2636-11e1-ad9e-0002a5d5c51b.elf] Error 1
make[2]: *** [package/pkg-generic.mk:284: /home/itaru/work/out-br/build/optee_test_ext-1.0/.stamp_built] Error 2
make[1]: *** [Makefile:23: _all] Error 2
make[1]: Leaving directory '/home/itaru/work/out-br'
make: *** [common.mk:341: buildroot] Error 2
```

jforissier, 2024-06-18:
Indeed, GCC 12 is incompatible with C++ TAs. One more thing to disable on M1. So `make -j$(nproc) RUST_ENABLE=n WITH_CXX_TESTS=n`.

jforissier, 2024-06-18:
@ikitayama please see https://github.com/OP-TEE/build/pull/764

ikitayama, 2024-06-18:
> Indeed, GCC 12 is incompatible with C++ TAs. One more thing to disable on M1. So `make -j$(nproc) RUST_ENABLE=n WITH_CXX_TESTS=n`.

Thanks! That worked.
```
make run-only
make[1]: Entering directory '/home/itaru/work/build'
ln -sf /home/itaru/work/build/../out-br/images/rootfs.cpio.gz /home/itaru/work/build/../out/bin/

* QEMU is now waiting to start the execution
* Start execution with either a 'c' followed by <enter> in the QEMU console or
* attach a debugger and continue from there.
*
* To run OP-TEE tests, use the xtest command in the 'Normal World' terminal
* Enter 'xtest -h' for help.

xterm: Xt error: Can't open display:
xterm: DISPLAY is not set
xterm: Xt error: Can't open display:
xterm: DISPLAY is not set
```

ikitayama, 2024-06-19:
> Indeed, GCC 12 is incompatible with C++ TAs. One more thing to disable on M1. So `make -j$(nproc) RUST_ENABLE=n WITH_CXX_TESTS=n`.

Requiring that buildroot's GCC be 11 leads to a `make toolchains` failure.

```

RUN curl https://storage.googleapis.com/git-repo-downloads/repo > /bin/repo && chmod a+x /bin/repo
RUN mkdir /optee
WORKDIR /optee
RUN repo init -u https://github.com/OP-TEE/manifest.git -m qemu_v8.xml && repo sync -j10
WORKDIR /optee/build
RUN make -j2 RUST_ENABLE=n WITH_CXX_TESTS=n toolchains
RUN make -j$(nproc) check RUST_ENABLE=n WITH_CXX_TESTS=n
```

jforissier, 2024-06-19:
It looks like the last Buildroot has lost support for GCC 11.x :-/ can you please try this patch:
```
diff --git a/br-ext/configs/sdk-common b/br-ext/configs/sdk-common
index 64e0b5c..18563be 100644
--- a/br-ext/configs/sdk-common
+++ b/br-ext/configs/sdk-common
@@ -1,6 +1,6 @@
 BR2_CCACHE=y
 BR2_CCACHE_USE_BASEDIR=y
-BR2_GCC_VERSION_11_X=y
+BR2_GCC_VERSION_14_X=y
 BR2_INIT_NONE=y
 BR2_KERNEL_HEADERS_6_1=y
 BR2_PACKAGE_BUSYBOX=n
```

It will likely not work due to  `BR2_KERNEL_HEADERS_6_1=y` being incorrect, but I do not know which version is expected for GCC 14.x, so please report any error message and we will take it from there. Thanks!

ikitayama, 2024-06-19:
> It looks like Buildroot has lost support for GCC 11.x :-/ can you please try this patch:
> 
> ```diff
> diff --git a/br-ext/configs/sdk-common b/br-ext/configs/sdk-common
> index 64e0b5c..18563be 100644
> --- a/br-ext/configs/sdk-common
> +++ b/br-ext/configs/sdk-common
> @@ -1,6 +1,6 @@
>  BR2_CCACHE=y
>  BR2_CCACHE_USE_BASEDIR=y
> -BR2_GCC_VERSION_11_X=y
> +BR2_GCC_VERSION_14_X=y
>  BR2_INIT_NONE=y
>  BR2_KERNEL_HEADERS_6_1=y
>  BR2_PACKAGE_BUSYBOX=n
> ```
> 
> It will likely not work due to `BR2_KERNEL_HEADERS_6_1=y` being incorrect, but I do not know which version is expected for GCC 14.x, so please report any error message and we will take it from there. Thanks!

@jforissier Will do that tomorrow in the morning my time!

ikitayama, 2024-06-20:
> It looks like Buildroot has lost support for GCC 11.x :-/ can you please try this patch:
>
> ```diff
> diff --git a/br-ext/configs/sdk-common b/br-ext/configs/sdk-common
> index 64e0b5c..18563be 100644
> --- [...]

It seems to be working? I am seeing two xterms saying "Secure World" and "Normal World" on my M1 Max. Builds were done in Ubuntu VM.

```
[...]
* QEMU is now waiting to start the execution
* Start execution with either a 'c' followed by <enter> in the QEMU console or
* attach a debugger and continue from there.
*
* To run OP-TEE tests, use the xtest command in the 'Normal World' terminal
* Enter 'xtest -h' for help.

Connection to 127.0.0.1 54320 port [tcp/*] succeeded!
Connection to 127.0.0.1 54321 port [tcp/*] succeeded!
cd /home/itaru/work/build/../out/bin && /home/itaru/work/build/../qemu/build/aarch64-softmmu/qemu-system-aarch64 \
	-nographic -smp 2 -cpu max,sme=on,pauth-impdef=on -d unimp -semihosting-config enable=on,target=native -m 1057 -bios bl1.bin -initrd rootfs.cpio.gz -kernel Image -append 'console=ttyAMA0,38400 keep_bootcon root=/dev/vda2 '  -object rng-random,filename=/dev/urandom,id=rng0 -device virtio-rng-pci,rng=rng0,max-bytes=1024,period=1000 -netdev user,id=vmnic -device virtio-net-device,netdev=vmnic -machine virt,acpi=off,secure=on,mte=off,gic-version=3,virtualization=false   -s -S -serial tcp:127.0.0.1:54320 -serial tcp:127.0.0.1:54321
QEMU 8.1.2 monitor - type 'help' for more information
(qemu)

ikitayama, 2024-06-20:
@jforissier I get to the prompt which I see on the Normal World xterm. Thanks a lot for your help!

jforissier, 2024-06-20:
> @jforissier I get to the prompt which I see on the Normal World xterm. Thanks a lot for your help!

Great news! I have updated https://github.com/OP-TEE/build/pull/764. Thanks!

closed 2024-07-27 by github-actions[bot]