OP-TEE/build #774: How to test and modify fTPM (enabled with buildroot package)

graziadonghia, 2024-08-09:
I am trying to implement a fTPM through tpm2-tools and tpm2-tss within OP-TEE. Firstly, I saw [this implementation](https://github.com/jbech-linaro/manifest/tree/ftpm?tab=readme-ov-file), but I could not build it because it was probably tested with an older version of OP-TEE. 
However, #480 has added support for fTPM through buildroot packages, so it should be included in the latest version of OP-TEE. Having said that, I have a couple of questions:

1) How can I build OP-TEE with buildroot and how can I further test it?
2) How can I execute measured boot and tpm functionalities with the new build?

jenswi-linaro, 2024-08-14:
#480 does if I understand it correctly only cover the rootfs part of fTPM. You still need to compile an fTPM TA for OP-TEE.

graziadonghia, 2024-08-28:
Is there a current implementation of an fTPM TA for the latest version of OP-TEE?

closed 2024-10-03 by github-actions[bot]