OP-TEE/build #775: Issues building for **rpi3** on Aarch64 Debian Bookworm.

msgilligan, 2024-08-12:
I was able to successfully build an SDCard image on Debian Bookworm (in a MacOS AArch64 VM) but ran into 3 minor issues:

1. The official instructions say to do `make -j8` followed by `make img-help`, but the output of `make img-help` is incorrect because `../out/rpi3-sdcard.img` has not been built yet. You must do a `make sdcard-image` to build that file.

2. When running `make sdcard-image` the following warning message is emitted:

```
../toolchains/aarch64/bin/aarch64-linux-ld.bfd: warning: trusted-firmware-a/build/rpi3/debug/bl1/bl1.elf has a LOAD segment with RWX permissions
```

I was able to work-around this error by making the following change in `trusted-firmware-a/Makefile:

```
-TF_LDFLAGS             +=      --fatal-warnings -O1
+TF_LDFLAGS             +=      --no-fatal-warnings -O1
```

There's probably a cleaner fix that resolves the warning and leaves warnings as fatal. But this workaround allows the image to be built and the resulting image successfully boots my Rpi3 and can run `xtest`.

3. I had to add `/usr/sbin` to my shell's `$PATH` or otherwise `make sdcard-image` would fail to find `fdisk` and `mkdosfs`.

jbech-linaro, 2024-08-12:
> 1. The official instructions say to do `make -j8` followed by `make img-help`, but the output of `make img-help` is incorrect because `../out/rpi3-sdcard.img` has not been built yet. You must do a `make sdcard-image` to build that file.

That implies that something went wrong with the overall build. You can definitely call `sdcard-image` by itself, but that target is called as a dependency from the [`all`](https://github.com/OP-TEE/build/blob/master/rpi3.mk#L69-L70) makefile rule. What I'm saying is that when the build is working as expected, there should be no need to explicitly call `sdcard-image`. I.e., we should fix the build errors and then this should just work. I.e., the instruction from the docs and `make img-help` are correct.

Personally I think it's overkill, but one could add to [`make img-help`](https://github.com/OP-TEE/build/blob/master/rpi3.mk#L206-L212) an if/else. I.e., pseudo `if sdcard image exists, $sudo dd if .... else echo SD-image hasn't been generated`. I'm not going to add that myself, but if someone else think it makes life easier for debugging build issues, than I'd be happy to accept a patch for it.

msgilligan, 2024-08-12:
> That implies that something went wrong with the overall build.

You are right. I must have not noticed that the overall build had failed. Now that (2) and (3) are fixed, the overall build is in fact building `rpi3-sdcard.img`.

closed 2024-09-18 by github-actions[bot]