OP-TEE/build #779: RPi .img file missing

aElizaF, 2024-09-19:
I'm following the Raspberry Pi build instructions. I've abandoned using a Rpi3, and am building on a Rpi5. I'm using Ubuntu 22.04. 
When using the make img-help in the build folder of the project directory, I get an error that the rpi3-sdcard.img does not exist. 

I received no errors when following the build instructions. I searched for the file in the project directory, and I searched for all .img files in the directory, with no success. 

I'm not sure what other troubleshooting I can do to get the image file. Any help would be great. Or perhaps a pointer to where I can download the rpi3 image, that way I can place it in the correct spot to complete set up. 

Thank you!

msgilligan, 2024-09-19:
Try `make sdcard-image`

See https://github.com/OP-TEE/build/issues/775#issuecomment-2283254845

aElizaF, 2024-09-20:
Awesome! Thank you! Closing this out.

closed 2024-09-20 by aElizaF