OP-TEE/build #789: Error building op-tee for QEMUv8

Toomay487, 2024-11-11:
Hello,

I am trying to build the OP-TEE project for QEMUv8. My build setup is with Ubuntu 20.04. I have followed the steps:

repo init -u https://github.com/OP-TEE/manifest.git -m qemu_v8.xml -b 3.22.0
repo sync -j4
make -f toolchain.mk toolchains
make -f qemu_v8.mk all
But during make run I am getting below error:

make[1]: Entering directory '/home/t87/optee/qemu'
changing dir to build for make ""...
make[2]: Entering directory '/home/t87/optee/qemu/build'
config-host.mak is out-of-date, running configure
bash: line 3: ./config.status: No such file or directory
make[2]: *** No rule to make target 'config-host.mak', needed by 'Makefile.prereqs'.  Stop.
make[2]: Leaving directory '/home/t87/optee/qemu/build'
make[1]: *** [GNUmakefile:11: all] Error 2
make[1]: Leaving directory '/home/t87/optee/qemu'
make: *** [Makefile:251: qemu] Error 2

Can anyone please let me know what is the issue here and how can I resolve it?

Thanks in advance.

jforissier, 2024-11-12:
Hi @Toomay487,

You may be missing some prerequisite package. Please refer to the Ubuntu 20.04 Dockerfile at https://optee.readthedocs.io/en/latest/building/prerequisites.html. Using that file and just adding `-b 3.22.0` to the `repo init` command, I could build everything in the container (`make -j$(nproc)`) and then run the tests (`make check-only`).

cinghioGithub, 2024-11-21:
Hi @Toomay487,

you could try to ```rm -rf ../qemu/build``` and build again. That solved the problem for me.

closed 2024-12-28 by github-actions[bot]