OP-TEE/build #816: Dependency issue between Optee-os and Optee-examples

MafiaStrange, 2025-02-15:
I am trying to build an embedded image with buildroot with optee enabled for my raspberry pi 4. I want to have optee-examples so that I could get to understand how CA interact with TAs in optee.

When I try adding optee-examples package through menuconfig, I get the error
_**optee-os is in the dependency chain of optee-examples that has added it to its _DEPENDENCIES variable without selecting it or depending on it from Config.in.**_

I have the line **_depends on BR2_TARGET_OPTEE_OS_** in packages/optee-examples/Config.in and the line **_OPTEE_EXAMPLES_DEPENDENCIES = optee-os optee-client_** in packages/optee-examples/optee-os.mk

Still I am not able to figure out where the issue is. I would be really grateful if you could help me in figuring it out

etienne-lms, 2025-03-18:
Hello @MafiaStrange,
Sorry I missed this Issue entry.

Indeed, Buildroot package **optee-examples** has a dependency on **optee-os**. This is because building the trusted applications (TAs) from OP-TEE Examples requires to build and install OP-TEE TA devkit (aka SDK, provided by **optee-os**) in the staging directory. This devkit provides the libraries and header files needed by TAs.
You should get a functional setup by enabling `BR2_TARGET_OPTEE_OS=y` and providing the OP-TEE OS minimal configuration information describing your platform (mainly telling it's ARM64 target) and stating you want only to build the OP-TEE devkit. I think I can assume RPI4 is Arm64 as RPI3 hence, using mainline source trees, something like the below should help:
```
BR2_TARGET_OPTEE_OS=y
BR2_PACKAGE_OPTEE_CLIENT=y
BR2_PACKAGE_OPTEE_EXAMPLE=y
BR2_TARGET_OPTEE_OS_PLATFORM=rpi3
BR2_TARGET_OPTEE_OS_CORE=n
BR2_TARGET_OPTEE_OS_SDK=y
```
You may want to set some specific debug configuration switches, like the below to get TAs debug trace message:
`BR2_TARGET_OPTEE_OS_ADDITIONAL_VARIABLES="CFG_TEE_TA_LOG_LEVEL=2"`

The above assumes your target embedded OP-TEE firmware embeds OP-TEE default key for signing TAs. If not, you need to provide the TA signing key with:
`BR2_TARGET_OPTEE_OS_ADDITIONAL_VARIABLES="... TA_SIGN_KEY=/path/to/TA-signing-key-pem-file"`

MafiaStrange, 2025-04-18:
It's working now.. thanks a lot!!!

closed 2025-05-25 by github-actions[bot]