OP-TEE/build #822: Missing license

gagachang, 2025-04-16:
All OP-TEE repositories have explicit license except `build` and `manifest`.
Why there is no explicit license for these two repositories?

jforissier, 2025-04-16:
Hi @gagachang, I'd say it is just an oversight. In my opinion, everything in there they should be licensed as BSD-2-Clause by default like `optee_os` etc. and we should add a LICENSE file. @jbech-linaro @jenswi-linaro any opinion?

jbech-linaro, 2025-04-16:
It should be `BSD-2-Clause` just as you said @jforissier .

jenswi-linaro, 2025-04-16:
Yes, I agree.

jforissier, 2025-04-16:
OK thanks for the feedback. @gagachang would you please submit pull requests to add the LICENSE file?

jforissier, 2025-04-16:
> > would you please submit pull requests to add the LICENSE file?
> 
> Sure, may I just copy the LICENSE file from optee_os ? https://github.com/OP-TEE/optee_os/blob/master/LICENSE

Yes, it's the proper one.

gagachang, 2025-04-16:
Please check https://github.com/OP-TEE/manifest/pull/318 for manifest.

closed 2025-04-16 by jforissier

gagachang, 2025-04-16:
@jforissier Will the two PRs be part of v4.6.0?

jforissier, 2025-04-16:
@gagachang yes, they will.