OP-TEE/build #839: Build error with QEMU on WSL-Ubuntu

Hussain1811, 2025-07-04:
**Steps:**
```
repo init -u https://github.com/OP-TEE/manifest.git -m qemu_v8.xml -b 4.6.0
repo sync -j4 --no-clone-bundle
cd build
make -j2 toolchains
make CFG_PKCS11_TA=y CFG_USER_TA_TARGET_pkcs11=ta_arm64 QEMU_VIRTFS_ENABLE=y -j 8 run
```

**Error Log:**
```
>>> host-acl 2.3.2 Extracting
mkdir -p /mnt/c/Users/SadiqHussainM/optee_qemu/out-br/per-package/host-acl/host
rsync -a --hard-links  --link-dest=/mnt/c/Users/SadiqHussainM/optee_qemu/out-br/per-package/host-tar/host/ /mnt/c/Users/SadiqHussainM/optee_qemu/out-br/per-package/host-tar/host/ /mnt/c/Users/SadiqHussainM/optee_qemu/out-br/per-package/host-acl/host
rsync: [generator] failed to hard-link /mnt/c/Users/SadiqHussainM/optee_qemu/out-br/per-package/host-tar/host/usr with usr: Is a directory (21)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
make[2]: *** [package/pkg-generic.mk:204: /mnt/c/Users/SadiqHussainM/optee_qemu/out-br/build/host-acl-2.3.2/.stamp_extracted] Error 23
make[1]: *** [Makefile:23: _all] Error 2
make[1]: Leaving directory '/mnt/c/Users/SadiqHussainM/optee_qemu/out-br'
make: *** [common.mk:354: buildroot] Error 2
make: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/mnt/c/Users/SadiqHussainM/optee_qemu/linux'
```

Hussain1811, 2025-07-04:
Would be great if i could get some help for this.
Thanks!

jenswi-linaro, 2025-07-07:
I was wondering if the FS has all the features needed to support buildroot.

closed 2025-08-13 by github-actions[bot]